AllStarLink/ASL3 #74: SIP connections not functioning, possibly due to requiring RTP code in pjsip

alexjj, 2024-06-30:
**Describe the bug**

I connect to my node on my computer via SIP. I've setup a VM running Debian 12 with ASL3, and trying to connect via SIP. I still have my existing node running to test/compare SIP config settings.

I can connect the SIP client (MicroSIP) to the asterisk server, and it successfully registers (as watching `asterisk -rvvv`. However, when I try to dial my node it gives the error:

```
ERROR[931][C-00000007]: rtp_engine.c:510 ast_rtp_instance_new: No RTP engine was found. Do you have one loaded?
NOTICE[931][C-00000007]: chan_sip.c:19667 send_check_user_failure_response: RTP init failure for device "GM5ALX" <sip:212@192.168.7.192>
```

The sip module is loaded, verified via `module show like sip` and my sip.conf and extensions.conf have the same SIP settings as my working node.

Searching the internet, leads to an [asterisk forum post](https://community.asterisk.org/t/cant-connect-to-asterisk-with-sip-client/93391/6) that suggests asterisk should be built with pjsip, as chan_sip is deprecated and some RTP code is within PJSIP.

I've been trying to step through all the config and read the commentary, in case there's something else new in ASL3 but so far haven't found anything. 

**Software versions (listed in asl-menu, option 4)**
 - asl3: [3.1.1-1]
 - asl3-asterisk: [2:20.8.1+asl3-3.0.0-1]

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - Debian 12, LXC, kernal 6.8.8-2

**Additional context**

Not at all familiar with Asterisk, and ended up using SIP with ASL2 so I could use my node whilst on the computer, and iaxrpt client seemed much worse than a SIP client.

W6HBR, 2024-06-30:
You definitely want to use PJSIP.  There is a section in the manual dedicated to SIP connections:  https://allstarlink.github.io/adv-topics/sip-phone/

alexjj, 2024-06-30:
Thanks for that. I did end up on the PJSIP docs on asterisk but didn't see that section of the docs - thanks for that 👍

Allan-N, 2024-06-30:
Please let us know if PJSIP works for you so we'll know that it's OK to close this issue.  And, if you have issues, tell us more.

alexjj, 2024-07-01:
PJSIP works great, thanks! I had a hiccup with my ASL3 install but then realised DKMS modules don't work on LXC without the host having them, so I skipped that plan. All good with connecting SIP client to the asterisk server and then my node!

closed 2024-07-01 by alexjj