AllStarLink/ASL3 #75: SimpleUSB settings lost on Asterisk restart

JayCue702, 2024-07-01:
AllStarLink3.0.10 Raspberry Pi image build

Using asl-menu > Node settings > Allstar Node Setup Menu > update node xxxxx > Interface Tune CLI:
I make my changes to RX level and TX levels, save changes, exit. I re-enter and settings are still active. If i restart asterisk or reboot, then RX/TX levels are back to default 500

re versions (listed in asl-menu, option 4)**
 - asl3: 3.0.0
 - asl3-asterisk: 20.8.1+asl3-3.0.0-1.deb12

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Raspberry Pi 3b+

Allan-N, 2024-07-02:
Can you check the filesystem permissions of `/etc/asterisk/simpleusb.conf` ?
Is the file owned by "asterisk" (correct) or "root" (incorrect) ?
If the latter, did you happen to edit this file yourself or using `asl-menu`?

JayCue702, 2024-07-02:
-rw-r--r-- 1 asterisk asterisk 5626 Jul  1 12:00 /etc/asterisk/simpleusb.conf

i used asl-menu to edit

Allan-N, 2024-07-02:
Re: asl-menu, I should have been more clear.  I wanted to know if you had used the "Expert Configuration Menu" to "Edit simpleusb.conf file" ?  But, the important thing to note is that the filesystem permissions look OK so that wasn't the source of your problem.

And, to be sure, let's try :
- Exec `tail /etc/asterisk/simpleusb.conf` and take node of the per-node settings
- `sudo asl-menu`
- 1 Node Settings
- 1 AllStar Node Setup Menu
- ? Update node #####
- 8 Interface Tune CLI
- Make "a" change (e.g. "Toggle RX Boost")
- "W" (to Write)
- "0" (to Exit)
- Back Back Back Back Exit
- Exec `tail /etc/asterisk/simpleusb.conf` and see if "rxboost" is present.

Now, if needed, you can repeat the above steps to revert whatever change you made.

JayCue702, 2024-07-02:
**initial tail results:**
; Note: the device string is automatically found when the
;       USB setting "devstr=" is empty.
;
; Note: the interface "tune" settings will be added to the
;       per-node settings (below).

[61841](node-main)
ctcssfrom = no
plfilter = no
devstr = 1-1.3:1.0

**Interface Tune CLI:**
Active Simple USB Radio device is [61841].
1) Select active USB device
2) Set Rx Voice Level (using display)
3) Set Transmit A Level (currently '800')
4) Set Transmit B Level (currently '800')
B) Toggle RX Boost (currently 'disabled')
C) Toggle Pre-emphasis (currently 'disabled')
D) Toggle De-emphasis (currently 'disabled')
E) Toggle Echo Mode (currently 'disabled')
F) Flash (Toggle PTT and Tone output several times)
G) Toggle PL Filter (currently 'disabled')
H) Toggle PTT mode (currently 'ground')
I) Change Carrier From (currently 'usbinvert')
J) Change CTCSS From (currently 'no')
K) Change RX On Delay (currently '0')
L) Change TX Off Delay (currently '0')
P) Print Current Parameter Values
S) Swap Current USB device with another USB device
T) Toggle Transmit Test Tone/Keying (currently 'disabled')
V) View COS, CTCSS and PTT Status
W) Write (Save) Current Parameter Values
0) Exit Menu

executed W then 0

**Final tail:**
; Note: the device string is automatically found when the
;       USB setting "devstr=" is empty.
;
; Note: the interface "tune" settings will be added to the
;       per-node settings (below).

[61841](node-main)
ctcssfrom = no
plfilter = no
devstr = 1-1.3:1.0

where does RX voice level get saved and transmit A and B levels?

JayCue702, 2024-07-02:
displays simpleusb. switched it and then back to hopefully reset it but same results.

JayCue702, 2024-07-02:
edited simpleusb.conf with above parameters you gave and after restart of asterisk, settings worked. dont know where it would save. Solution though is maybe have simpleusb.conf have default values of those parameters in it before, because now it updates changes to file.

Allan-N, 2024-07-02:
Odd.  The simpleusb.conf file should have started off with:
```
...
[1999](node-main)

;;;;; ASL3 Tune settings ;;;;;
devstr=
rxmixerset=500
txmixaset=500
txmixbset=500
```
with [1999] being updated when you used the menu to change the node # and the devstr being updated on your first "Write".

One tidbit with the template-ized configuration files.  If you set a variable in a child category (e.g. a node) to a value that matches that of its parent template then the variable will be removed from the child (no need to have duplicate content in the file).  This confused us for a bit until we understood the behavior.

All good now?

closed 2024-07-02 by JayCue702