AllStarLink/ASL3 #76: Tx gets stuck on during bootup...

w8cpt, 2024-07-02:
Pi 3 A+ and a AURSINC Allstar RF Node UHF SR-FRS Radio Programming Module Pi-hat for Allstar hangs up and leaves TX on when I shut down the node properly after the install and reboot it. Here's a video and my buddies Shari Pi-hat did the same thing. He put a 30 second delay in Asterisk an entry in system d service file and it seemed to correct it...I don't know how to do that!
https://youtu.be/mnH9BxGZWqQ?si=lEwRawGfE0p0Ch5a

**Have you run a software update and rebooted?**
 - YES 

**Pi 3 A+

jxmx, 2024-07-02:
I'm a bit confused by this report. You're talking about a PiHat but linking to a USB-attached appliance board. To confirm, this is what you're trying to use, correct? https://www.amazon.com/AURSINC-Programming-Echolink-Compatible-Raspberry/dp/B0BNKS4GGH

JayCue702, 2024-07-02:
heres quick solution for delay. in terminal

sudo nano /lib/systemd/system/asterisk.service

then add following line above ExecStart:
ExecStartPost=/bin/sleep 30

then restart asterisk and see if that worked

jxmx, 2024-07-02:
@JayCue702 - You do not want to do that. `/lib/systemd/system/asterisk.service` will get blown away on the next upgrade. If ytou really want to change the systemd unit, then the correct procedure is to copy the `/lib/systemd/system` unit to `/etc/systemd/system`, edit the file there, and then do a `systemctl daemon-reload`.

jxmx, 2024-07-02:
Philosophically though, adding a sleep to `ExecStartPre` is really a bandge over a different problem. If this is only happening at boot, then most likely this unit is very slow to initialize and/or requires some other system process to be available before asterisk tries to use it.

tsawyer, 2024-07-02:
One of my testers is having this issue. I think we need to find the root cause and a solution.

jxmx, 2024-07-02:
We either need someone on the development team who has the hardware that’s a problem or get SSH and root access to someone’s note that is having the problem.

KB4MDD, 2024-07-02:
As @jxmx noted, we need hardware to test or dedicated access to a computer.

jxmx, 2024-07-02:
Tim what hardware is your tester seeing the issue on?

W6HBR, 2024-07-02:
> One of my testers is having this issue. I think we need to find the root cause and a solution. Any debugging suggestions?

Tim, I have a HotSpotRadio that exhibits this issue.  First boot keys up and stays.  If I go into the tune menu and update settings, then restart Asterisk, it stops.  Unfortunately, I do not have time to debug right now, but you're welcome to come by and grab the unit if you wish.

jxmx, 2024-07-02:
@W6HBR what model?

W6HBR, 2024-07-02:
Pi 3 w/ hat

tsawyer, 2024-07-02:
> Tim what hardware is your tester seeing the issue on?

@jxmx Rpi3b with DMK URIxb.

JayCue702, 2024-07-02:
I was having issues getting simpleusb.conf settings to save on reboot. Could this be possibility for this issue? Does it need usbinvert?

w8cpt, 2024-07-02:
Yes that's what I'm using. My buddy has the same issue with a Shari...

JayCue702, 2024-07-02:
Post your tail

tail /etc/asterisk/simpleusb.conf

w8cpt, 2024-07-02:
I don't know what that means...can you explain more?

JayCue702, 2024-07-02:
Go to terminal and type that line in above then copy and paste what it says here

w8cpt, 2024-07-02:
asl@node588840:~ $ tail /etc/asterisk/simpleusb.conf

[588840](node-main)

ctcssfrom = no

;;;;; ASL3 Tune settings ;;;;;
devstr = 1-1:1.0
rxmixerset = 650
txmixaset = 650
txmixbset = 650

w8cpt, 2024-07-02:
anything else while im at the terminal?

w8cpt, 2024-07-02:
> I'm a bit confused by this report. You're talking about a PiHat but linking to a video of a USB-attached appliance board. To confirm, this is what you're trying to use, correct? https://www.amazon.com/AURSINC-Programming-Echolink-Compatible-Raspberry/dp/B0BNKS4GGH

yes thats what im using Amazon called it that I copied it from Amazon...

JayCue702, 2024-07-02:
Not sure if all these hats are equal but can you try adding line to end of simpleusb.conf

carrierfrom = usb

w8cpt, 2024-07-02:
> Not sure if all these hats are equal but can you try adding line to end of simpleusb.conf
> 
> carrierfrom = usb

ive tried that in the tune menu on the asl-menu and it still does it....

JayCue702, 2024-07-02:
I had issues with parameters not saving using tune menu. Go to expert settings and edit simpleusb.conf. I'm just checking if you had same issue I had

w8cpt, 2024-07-02:
anything else before I shut down this node?

w8cpt, 2024-07-02:
> I had issues with parameters not saving using tune menu. Go to expert settings and edit simpleusb.conf. I'm just checking if you had same issue I had

that happend to me once but the way its currently configured it saves the settings. not sure what I changed other than when it didn't save the USB settings the "Restart Asterisk" popped up. when it actually saved the settings i went and restarted asterisk myself in the menu...

w8cpt, 2024-07-02:
here's my settings. i have tried "I" and "J" no, USB and USBINVERT and I still had the restart issue...

Active Simple USB Radio device is [588840].
1) Select active USB device
2) Set Rx Voice Level (using display)
3) Set Transmit A Level (currently '650')
4) Set Transmit B Level (currently '650')
B) Toggle RX Boost (currently 'disabled')
C) Toggle Pre-emphasis (currently 'disabled')
D) Toggle De-emphasis (currently 'disabled')
E) Toggle Echo Mode (currently 'disabled')
F) Flash (Toggle PTT and Tone output several times)
G) Toggle PL Filter (currently 'enabled')
H) Toggle PTT mode (currently 'ground')
I) Change Carrier From (currently 'usbinvert')
J) Change CTCSS From (currently 'no')
K) Change RX On Delay (currently '0')
L) Change TX Off Delay (currently '0')
P) Print Current Parameter Values
S) Swap Current USB device with another USB device
T) Toggle Transmit Test Tone/Keying (currently 'disabled')
V) View COS, CTCSS and PTT Status
W) Write (Save) Current Parameter Values
0) Exit Menu

Please enter your selection now:

JayCue702, 2024-07-02:
thats because its not saving like i said using that menu and when you did that tail command it shows its not saving. so i need you to go
sudo asl-menu > expert configuration menu > edit simpleusb.conf file

scroll all the way to bottom and add line
carrierfrom = usb

press ctrl x,
press y
press enter

now restart your node

JayCue702, 2024-07-02:
not sure if it works for your device but this worked for someone elses setup. seems to be few different hats on the market

jxmx, 2024-07-02:
So some of these adapters are COR=high? Is that what we're saying?

w8cpt, 2024-07-02:
> thats because its not saving like i said using that menu and when you did that tail command it shows its not saving. so i need you to go sudo asl-menu > expert configuration menu > edit [...]

But it says  650 and 650 so it saved them...

tsawyer, 2024-07-03:
From what I can gather, the transmitter gets stuck only on first boot from a new microSB. This might be related to a kernel upgrade during the boot process. 

Here are separate servers where  USB dies on boot up. I believe these are both Rpi 3b. See snips from /var/log/asterisk/messages.log from each server:

This is HopSpotRadios node...
```
[2024-03-15 08:10:32.501] NOTICE[1100] app_rpt.c: Normal Repeater Init  1999
[2024-03-15 08:10:32.503] NOTICE[1105] chan_simpleusb.c: Channel 1999: Automatically assigned USB device 1-1.4:1.0 to SimpleUSB channel
[2024-03-15 08:10:32.507] NOTICE[1104] chan_simpleusb.c: Channel 1999: Set option TONE VERIFY, mode: OFF(0).
[2024-03-15 08:10:32.562] WARNING[1104] chan_simpleusb.c: Possibly stuck USB read channel. [1999]
[2024-03-15 08:10:32.587] WARNING[1104] chan_simpleusb.c: USB read channel [1999] was not stuck.
[2024-07-02 19:58:06.210] ERROR[1104] chan_simpleusb.c: Channel 1999: HID process has died or is not responding.
[2024-07-02 19:58:10.000] WARNING[1100] app_rpt.c: rpt_thread restarted on node 1999
[2024-07-02 19:58:10.011] NOTICE[1726] chan_simpleusb.c: Channel 1999: Set option TONE VERIFY, mode: OFF(0).
[2024-07-02 19:58:10.013] ERROR[1727] chan_simpleusb.c: Channel 1999: Unable to open DSP device 1: Device or resource busy.
[2024-07-02 19:58:14.003] ERROR[1726] chan_simpleusb.c: Channel 1999: HID process has died or is not responding.
[2024-07-02 20:03:02.328] NOTICE[1081] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 162.248.92.131:443 to 34.105.111.212:443
``` 

And this is the other server. It has a URI...
```
[2024-06-30 14:56:02.912] NOTICE[1024] app_rpt.c: Normal Repeater Init  47050
[2024-06-30 14:56:02.915] NOTICE[1028] chan_simpleusb.c: Channel 47050: Set option TONE VERIFY, mode: OFF(0).
[2024-06-30 14:56:36.695] ERROR[1028] chan_simpleusb.c: Channel 47050: HID process has died or is not responding.
[2024-06-30 14:56:37.127] WARNING[1024] app_rpt.c: rpt_thread restarted on node 47050
[2024-06-30 14:56:37.130] NOTICE[1281] chan_simpleusb.c: Channel 47050: Set option TONE VERIFY, mode: OFF(0).
[2024-06-30 14:56:37.160] ERROR[1282] chan_simpleusb.c: Channel 47050: Unable to open DSP device 1: Device or resource busy.
[2024-06-30 14:56:37.182] WARNING[1281] chan_simpleusb.c: Possibly stuck USB read channel. [47050]
[2024-06-30 14:56:37.207] WARNING[1281] chan_simpleusb.c: USB read channel [47050] was not stuck.
[2024-06-30 14:56:41.005] ERROR[1281] chan_simpleusb.c: Channel 47050: HID process has died or is not responding.
[2024-06-30 15:01:18.059] NOTICE[997] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 162.248.92.131:443 to 34.105.111.212:443
```
After a second reboot, all is well.
```
[2024-06-30 15:05:21.535] NOTICE[1882] app_rpt.c: Normal Repeater Init  47050
[2024-06-30 15:05:21.538] NOTICE[1885] chan_simpleusb.c: Channel 47050: Set option TONE VERIFY, mode: OFF(0).
[2024-06-30 15:05:21.614] WARNING[1885] chan_simpleusb.c: Possibly stuck USB read channel. [47050]
[2024-06-30 15:05:21.638] WARNING[1885] chan_simpleusb.c: USB read channel [47050] was not stuck.
[2024-06-30 15:06:08.804] DTMF[1885] channel.c: DTMF begin '*' received on SimpleUSB/47050
[2024-06-30 15:06:08.804] DTMF[1885] channel.c: DTMF begin passthrough '*' on SimpleUSB/47050
```

w8cpt, 2024-07-03:
It's not just the 1st boot on a new SD card. It happens every boot for me....

tsawyer, 2024-07-03:
Does your log show the use is stuck?

w8cpt, 2024-07-03:
I don't know what log to look at. If you show me I will be able to copy it Saturday if someone doesn't figure it out before then. Mine does it every boot up now  it keys up and get stuck on tx and if I unplug the Aursinc and plug it back in it's not keyed up anymore but I can't TX or RX...my buddies Shari does the same on every boot up.

I'll attach a photo of something someone on Facebook said fixes it but I can't mess with it til Saturday so I don't know if it works...

He said " Don McJr  because there is no stanza to drop the ground for ptt. That's why adding this line forces it to pull the ground hence stopping the PTT. All of this software is new and still in BETA! Take everything with open arms. But what do I know.

I can also tell you how to turn off  cwid but I'm probably wrong too."
[image]

jxmx, 2024-07-03:
For anyone having problems with the "TX on boot", I'd like people to try something. As root/via sudo, please do the following:

```
rm -f /etc/systems/system/asterisk.service 
wget -O /lib/systemd/system/asterisk.service https://repo.allstarlink.org/hotfix/asl3_76_asterisk_systemd.service
systemctl daemon-reload
systemctl restart asterisk
```

After this, please test your device **WITHOUT** rebooting and confirm everything is okay. Then reboot the unit and see if the hung TX happens.

tsawyer, 2024-07-03:
The new service did not work for my friend with the URI.

Tx locks up on every reboot (he wasn't clear that it was every reboot previously). Can be cleared with an Asterisk restart each time.

KB4MDD, 2024-07-03:
It might help to get the simpleusb.conf and messages.log to review.

jxmx, 2024-07-03:
It has been verified that RTC/wallclock/hwclock interactions are causing this at boot. Linked the issue in app_rtp above. Short term fix is to apply:

```
rm -f /etc/systemd/system/asterisk.service 
wget -O /lib/systemd/system/asterisk.service https://repo.allstarlink.org/hotfix/asl3_76_asterisk_systemd.service
systemctl daemon-reload
```

**Note: this is not a permanent solution and may cause long delays in startup**

Having the asterisk start depending on `time-sync.target` can delay asterisk starting for anywhere from 30 seconds to minutes depending on how quickly timesyncd is able to set the clock accurately. This will also fail to ever launch with a node is not a network with a viable Internet connection to set the clock. So this is a very temporary workaround.

mpannen1979, 2024-07-04:
A pi4 with SimpleUSB device locked on TX for me once, and both that unit and a pi3 with SimpleUSB devices had error messages in asterisk log file regarding 'USB HID' audio device lock ups. If I restarted asterisk things were OK.

I am trying the 'delay startup of asterisk for time sync' in a service override file as mentioned above.

Both the pi3 and pi4, after the startup delay for time sync, worked on the first boot, no USB HID error messages listed in asterisk log file.

saffrey, 2024-07-06:
> For anyone having problems with the "TX on boot", I'd like people to try something. As root/via sudo, please do the following:
> 
> ```
> rm -f /etc/systemd/system/asterisk.service 
> wget -O /lib/systemd/system/asterisk.service https://repo.allstarlink.org/hotfix/asl3_76_asterisk_systemd.service
> systemctl daemon-reload
> systemctl restart asterisk
> ```
> 
> After this, please test your device **WITHOUT** rebooting and confirm everything is okay. Then reboot the unit and see if the hung TX happens.

Can confirm that this worked for me. Pi3 with SA818 via USB. Thanks!

jxmx, 2024-07-06:
The proposed fix will be included in 3.0.1

mpannen1979, 2024-07-07:
the 3.0.1 packages from devel seem to do the trick for me

tsawyer, 2024-07-07:
The root issue was on Raspberry Pi nodes without a realtime clock. A large time shift while Asterisk was running would cause Asterisk to become unresponsive and stop the usb channel drivers among other problems. Depending on the timing the PTT would lock asserted. Some nodes would lock in Tx on every boot, others only the during first boot after an install, or never lock in PTT and only appear unresponsive. Restarting Asterisk would always "fix" the problem until the next reboot.

jxmx, 2024-07-07:
This fix will now likely be included in the 3.0.2 production release. There is an issue that needs to be resolved with 3.0.1 before production release. New devel packages coming later today.

N0NB, 2024-07-07:
On a Pi 3B+ with ASL3 using a Techno By George FOB with the Simple USB driver, the FOB appeared non-responsive to COS.  An Asterisk restart brought to life.  So far I've not seen the PTT lockup but I've not rebooted much in my testing this past week.  At least now that I'm aware I can anticipate it until the update hits the repository.

jxmx, 2024-07-10:
Fixed in ASL3-Asterisk v20.8.1+asl3-3.02

w8cpt, 2024-07-10:
So there's an update we can do or do we have to download a new image version?

jxmx, 2024-07-10:
https://community.allstarlink.org/t/asl3-3-02-release-notes/21364

w8cpt, 2024-07-10:
> It appears to me they put that version in the main updates now, so a
> regular apt update, apt upgrade should do it?

OK thanks!

tsawyer, 2024-07-10:
This should also fix new node installs as update and upgrade are part of the installation process.

tsawyer, 2024-07-10:
I burned a 3.0.10 again to test this to test my previous assumption. It appears to be correct. However, it seemed to take a very long time to install.

It keyed up and sent a courtesy tone. I think it dropped briefly. Then it keyed up again and stay keyed for maybe 3 to 5 minutes. Seemed like forever. There was disk activity the whole time so I patiently waited. It finally sent a second courtesy tone and IDed. 

I can only assume the long wait after the first courtesy tone was the Pi doing the asl3-3.0.2 updates. Given that is the case perhaps a new Pi image is warranted to dissuade user anxiety and prevent premature install aborts. 

Opening this issue again for further consideration.

w8cpt, 2024-07-11:
3 to 5 minutes of keyed up is not good so yeah still needs to be addressed

jxmx, 2024-07-11:
Image 3.0.11 spun and published.

closed 2024-07-11 by jxmx

tsawyer, 2024-07-11:
Install from 3.0.11 Pi image on a HotSpotRadios half duplex node went fine. Activity LED started blinking about 2 ½ minutes after power up, tx keyup, courtesy tone, ID and Tx unkey.

toyotahead, 2024-11-04:
I am a bit late to this party but I too am experiencing very similar issues with the PTT being stuck keyed upon first boot. However restarting asterisk doesn't solve it for me. To make it behave I was in Interface Tune CLI, and entered F to make the transmitter key up a few times. (obviously the PTT is already keyed) Upon doing so the 3 tones can be heard, and once done the PTT is off, and from then on keys up normally until the next reboot.

I am somewhat fresh to ASL3 but am reasonably competent in terminal with some direction. I have also tried the 2 suggestions above with no success. 

My setup is a fresh debian 12 install in a proxmox (v8.2.7) vm with a CM108 passed through from following these instructions: https://allstarlink.github.io/user-guide/install/  
Using SimpleUSB configuration as full duplex into dual CDM750's

                                                         OS            : Debian GNU/Linux 12 (bookworm)
                                                      │ OS Kernel     : 6.1.0-26-amd64
                                                      │
                                                      │ Asterisk      : 20.9.3+asl3-3.0.6-1.deb12
                                                      │ ASL [app_rpt] : 3.0.6
                                                      │
                                                      │ Installed ASL packages :
                                                      │
                                                      │   Package                         Version
                                                      │   ==============================  
                                                      │   allmon3                         1.4.1-1.deb12
                                                      │   asl3                            3.5.0-1.deb
                                                      │   asl3-asterisk                   2:20.9.3+asl3-3.0.6-1.deb12
                                                      │   asl3-asterisk-config            2:20.9.3+asl3-3.0.6-1.deb12
                                                      │   asl3-asterisk-modules           2:20.9.3+asl3-3.0.6-1.deb12
                                                      │   asl3-menu                       1.9-1.deb12
                                                      |    dahdi                           1:3.1.0-2
                                                      │   dahdi-dkms                      1:3.4.0-4+asl
                                                      │   dahdi-linux                     1:3.4.0-4+asl

/var/log/asterisk/messages.log    from the last boot up
`
[2024-11-04 13:48:55.291] Asterisk 20.9.3+asl3-3.0.6-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-10-05 13:19:13 UTC
[2024-11-04 13:48:55.348] NOTICE[481] loader.c: 71 modules will be loaded.
[2024-11-04 13:48:55.592] NOTICE[481] cdr.c: CDR simple logging enabled.
[2024-11-04 13:48:55.596] NOTICE[481] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2024-11-04 13:48:55.624] NOTICE[481] indications.c: Default country for indication tones: us
[2024-11-04 13:48:55.624] NOTICE[481] indications.c: Setting default indication country to 'us'
[2024-11-04 13:48:55.746] NOTICE[481] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2024-11-04 13:48:55.761] WARNING[481] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-11-04 13:48:55.761] WARNING[481] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-11-04 13:48:55.761] NOTICE[481] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 23.
[2024-11-04 13:48:55.761] NOTICE[481] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 31.
[2024-11-04 13:48:55.761] NOTICE[481] chan_dahdi.c: Ignoring any changes to 'hassip' (on reload) at line 35.
[2024-11-04 13:48:55.761] NOTICE[481] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 39.
[2024-11-04 13:48:55.761] NOTICE[481] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
[2024-11-04 13:48:55.761] WARNING[481] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2024-11-04 13:48:55.761] WARNING[481] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2024-11-04 13:48:56.096] WARNING[481] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2024-11-04 13:48:56.323] NOTICE[644] app_rpt.c: Normal Repeater Init  53011
[2024-11-04 13:48:56.329] NOTICE[652] chan_simpleusb.c: Channel 53011: Set option TONE VERIFY, mode: OFF(0).
[2024-11-04 13:48:56.352] ERROR[653] chan_simpleusb.c: Channel 53011: Device string 2-1:1.0 was not found.
[2024-11-04 13:53:04.644] Asterisk 20.9.3+asl3-3.0.6-1.deb12 built by builder @ allstarlink.org on a x86_64 running Linux on 2024-10-05 13:19:13 UTC
[2024-11-04 13:53:04.694] NOTICE[491] loader.c: 71 modules will be loaded.
[2024-11-04 13:53:04.939] NOTICE[491] cdr.c: CDR simple logging enabled.
[2024-11-04 13:53:04.943] NOTICE[491] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2024-11-04 13:53:04.971] NOTICE[491] indications.c: Default country for indication tones: us
[2024-11-04 13:53:04.971] NOTICE[491] indications.c: Setting default indication country to 'us'
[... cut ...]
[2024-11-04 14:04:27.736] NOTICE[615] app_rpt.c: Normal Repeater Init  53011
[2024-11-04 14:04:27.741] NOTICE[624] chan_simpleusb.c: Channel 53011: Set option TONE VERIFY, mode: OFF(0).
[2024-11-04 14:09:27.284] NOTICE[596] dnsmgr.c: dnssrv: host 'register.allstarlink.org' changed from 34.105.111.212:443 to 162.248.92.131:443

`

Like i said it seems to work great until the VM is rebooted.  I am not sure why it hangs the PTT during the initial boot up.....  Sorry for bringing this up again guys....

Thoughts?

Allan-N, 2024-11-04:
@toyotahead This GitHub issue was closed a while back.  Also, since your software/hardware configuration may be different it could be unrelated to what was reported here.  Please file a NEW GitHub issue for the problems you are experiencing ... and please include a bit more detail about YOUR configuration (e.g. what specific CM108 adapter you are using, your simpleusb.conf file, etc.).