AllStarLink/ASL3 #77: Unable to connect to EchoLink node

jetski28, 2024-07-02:
**Describe the bug**
Cannot connect to EchoLink node and get the following error in the logs...
WARNING[1215][C-00000012] pbx.c: Channel 'echolink/el0-4' sent to invalid extension but no invalid handler: context,exten,priority=radio-secure,55,1

**To Reproduce**
Steps to reproduce the behavior:
1. Configure rpt.conf with node 55 instead of 1999
2. Configure echolink.conf with our Echolink details
3. Configure extensions.conf with NODE = 55
4. Restart ASL

**Expected behavior**
External users to be able to connect to Echolink node

**Screenshots**
N/A

**Software versions (listed in asl-menu, option 4)**
 - asl3: 3.1.1-1.deb
 - asl3-asterisk: 2:20.8.1+asl3-3.0.0-1.deb12

**Have you run a software update and rebooted?**
YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Debian 12 on Proxmox

**Additional context**
I compared configuration with our working ASL2 node.
As a workaround, I replaced [radio-secure] configuration under extensions.conf with the following one line and now users can connect to Echolink node. It seems something with the default [radio-secure] configuration is preventing echolink conncetions.
`exten => ${NODE},1,rpt,${NODE}`

tsawyer, 2024-07-02:
Node number 55 is not a valid node number. The current usage of private node numbers should be at least 4 digits and start with 1. Please try another node number perhaps 1055 and test.

jetski28, 2024-07-08:
I see, moved it to 1055 and appears to be working.

closed 2024-07-08 by jetski28