AllStarLink/ASL3 #78: Echolink - No node information

jetski28, 2024-07-02:
**Describe the bug**
I feel like this might be an ASL issue as it appears to happen on both Supermon & Allmon3. Supermon shows Echolink connections with "No info" and Allmon3 shows "No database information". The Asterisk CLI command `echolink show nodes` shows the correct echolink information.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to node via Echolink
2. View Supermon or Allmon 
3. Look at Echolink connections
4. See no echolink information

**Expected behavior**
See the correct callsign of echolink connections in dashboard

**Screenshots**
[image]
[image]

**Software versions (listed in asl-menu, option 4)**
 - asl3: 3.1.1-1.deb
 - asl3-asterisk: 2:20.8.1+asl3-3.0.0-1.deb12

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - Debian 12 on Proxmox

**Additional context**
This has been a problem with ASL2 as well.

jxmx, 2024-07-02:
Can you provide the output you mentioned for `echolink show nodes`? Unfortunately, processing that output is pretty hairy because there's no data structure to it and likely something in your echolink info is unparsable.

jetski28, 2024-07-03:
[image]

KB4MDD, 2024-07-03:
It appears that your node is not downloading the master node list.

At the cli type `core set debug 7 chan_echolink.so`

Every 2 minutes chan_echolink will attempt to download the directory.  If there is a problem, you should see it on the console.

I will note that echolink.conf should have these entries:

```
; Max 4 servers
server1 = nasouth.echolink.org
server2 = naeast.echolink.org
server3 = servers.echolink.org
server4 = backup.echolink.org
```

jetski28, 2024-07-08:
Servers are correct.

No database download errors but when I connect I get ... 
`ERROR[11283]: chan_echolink.c:3219 do_new_call: Cannot find database entry for IP address x.x.x.x, Callsign <CALLSIGN RETRACTED>.`

If I disconnect after a couple minutes and reconnect the error doesn't appear again, assuming because the database has updated with my callsign?

But still Allmon3 and Supermon doesn't show any connection info.

KB4MDD, 2024-07-08:
Are you using the echolink client to connect to your allstar node?

If you wait 2 minutes for the node update, do you still have that problem?

Run the cli command `echolink dbget callsign xxxxxx`   enter your call sign for xxxxx in upper case and see what it returns.

jetski28, 2024-07-19:
I'm using the Echolink client on my iPhone.

I've let the app connect, then a couple minutes later, I've checked to see if my callsign has populated with the 'echolink dbget' command, once that shows correctly I've then connected, and still shows No Database Info or No Info on both Allmon3 and Supermon.

jxmx, 2024-07-21:
Please exactly copy/paste the output of `echolink dbget nodename ECHOLINKID`. Echolink node numbers are supposed to be 6 digit numbers starting with a "3". Your starts with a six. Allmon3 won't try to look that up as an Echolink node. I'm assuming Supermon does the same thing but I don't use it.

jxmx, 2024-07-21:
After you connect will you please copy/paste the output of the following commands here replacing "NODE" with your node number and ECHOLINK_ID with your digit echolink ID that starts with a 3.
```
rpt xnode NODE
rpt nodes NODE
echolink dbget nodename ECHOLINK_ID
```

jetski28, 2024-07-22:
`ASL-Echolink*CLI> rpt xnode 1055
43129     192.168.8.175       1           OUT        65:59:03            ESTABLISHED         ~3224805                       1           IN         05:27:42            ESTABLISHED         ~3661351                       1           IN         00:00:13            ESTABLISHED         ~

T20, T25, T3001262, T3224805, T3661351, T430130, T43129, T45472, T458982, T48798, T49276, T50522, T508880, T510640, T513020, T521600, T521603, T521604, T521605, T521606, T521607, T521609, T52748, T53390, T53930, T545441, T547452, T550114, T55767, T565700, T565701, T570070, T589545, T59964, T60520, T61148, T61252, T61488, T61943, T622430, TVK3NCR

RPT_TXKEYED=0
RPT_NUMLINKS=41
RPT_LINKS=41,T43129,T50522,T53930,T3001262,T622430,T45472,T52748,T550114,T61148,T49276,T521603,T20,T521609,T458982,T521607,T53390,T545441,T55767,T430130,T59964,T48798,T61943,T510640,T547452,T570070,T508880,T60520,T513020,T61488,T589545,T25,T565700,TVK3NCR,T521605,T521600,T61252,T565701,T521604,T521606,T3661351,T3224805
RPT_NUMALINKS=3
RPT_ALINKS=3,43129TU,3661351TU,3224805TU
RPT_ETXKEYED=0
RPT_AUTOPATCHUP=0
RPT_RXKEYED=0

parrot_ena=0
sys_ena=1
tot_ena=1
link_ena=1
patch_ena=1
patch_state=4
sch_ena=1
user_funs=1
tail_type=0
iconns=1
tot_state=2
ider_state=1
tel_mode=2

ASL-Echolink*CLI> rpt nodes 1055

************************* CONNECTED NODES *************************

T20, T25, T3001262, T3224805, T3661351, T430130, T43129, T45472
T458982, T48798, T49276, T50522, T508880, T510640, T513020, T521600
T521603, T521604, T521605, T521606, T521607, T521609, T52748, T53390
T53930, T545441, T547452, T550114, T55767, T565700, T565701, T570070
T589545, T59964, T60520, T61148, T61252, T61488, T61943, T622430
TVK3NCR

ASL-Echolink*CLI> echolink dbget nodename 661351
661351|VK3APJ|x.x.x.x (IP REDACTED)`

jxmx, 2024-09-19:
That all looks correct. Can you provide the output of `rpt xstat 1055`? Also, what version of  asterisk are you currently running?

jetski28, 2024-09-20:
rpt xstat doesn't appear to be a command.

However just in case your refering to xnode or stats, see below.
`ASL-Echolink*CLI> rpt stats 1055
************************ NODE 1055 STATISTICS *************************

Selected system state............................: 0
Signal on input..................................: NO
System...........................................: ENABLED
Parrot Mode......................................: DISABLED
Scheduler........................................: ENABLED
Tail Time........................................: STANDARD
Time out timer...................................: ENABLED
Incoming connections.............................: ENABLED
Time out timer state.............................: RESET
Time outs since system initialization............: 2
Identifier state.................................: CLEAN
Kerchunks today..................................: 0
Kerchunks since system initialization............: 0
Keyups today.....................................: 517
Keyups since system initialization...............: 10973
DTMF commands today..............................: 0
DTMF commands since system initialization........: 1
Last DTMF command executed.......................: 81343129
TX time today....................................: 02:54:52:94
TX time since system initialization..............: 73:48:28:877
Uptime...........................................: 553:57:18
Nodes currently connected to us..................: 43129, 3661351
Autopatch........................................: ENABLED
Autopatch state..................................: DOWN
Autopatch called number..........................: N/A
Reverse patch/IAXRPT connected...................: DOWN
User linking commands............................: ENABLED
User functions...................................: ENABLED

ASL-Echolink*CLI> rpt xnode 1055
43129     192.168.8.175       0           OUT        539:14:42           ESTABLISHED         ~3661351                       1           IN         00:01:38            ESTABLISHED         ~

T20, T28467, T3661351, T43129, T458982, T48798, T50522, T521600, T521601, T521603, T521604, T521605, T521606, T521607, T521609, T526610, T52697, T53390, T53930, T545441, T550114, T565700, T565701, T570070, T589545, T594540, T603781, T60754, T61935, T62375, T62657, TVK4GCR

RPT_NUMLINKS=32
RPT_LINKS=32,T43129,T50522,T28467,T62375,T545441,TVK4GCR,T521600,T62657,T550114,T20,T521603,T521609,T53390,T60754,T521607,T589545,T458982,T61935,T526610,T53930,T521605,T521601,T521604,T521606,T603781,T565701,T570070,T594540,T565700,T48798,T52697,T3661351
RPT_NUMALINKS=2
RPT_ALINKS=2,43129TU,3661351TU
RPT_TXKEYED=0
RPT_ETXKEYED=0
RPT_AUTOPATCHUP=0
RPT_RXKEYED=0

parrot_ena=0
sys_ena=1
tot_ena=1
link_ena=1
patch_ena=1
patch_state=4
sch_ena=1
user_funs=1
tail_type=0
iconns=1
tot_state=2
ider_state=2
tel_mode=2
`

Version
`OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-22-amd64

Asterisk      : 20.8.1+asl3-3.0.0-1.deb12
ASL [app_rpt] : 3.0.0

allmon3               1.2.1-2
asl3                  3.1.1-1.deb
asl3-asterisk         2:20.8.1+asl3-3.0.0-1.deb12
asl3-asterisk-config  2:20.8.1+asl3-3.0.0-1.deb12
asl3-asterisk-module  2:20.8.1+asl3-3.0.0-1.deb12
asl3-menu             1.6-1.deb12
`

jxmx, 2024-09-20:
I meant `xstat` sorry. Also, will you please update to the latest packages so we can make sure we're looking at production here? You're pretty far behind the current code.

jxmx, 2024-09-20:
No, I need the output of "XStat" which I forgot isn't in the CLI, only in the AMI. This is something unique to your setup and I'm not sure why it's happening. If you make an Echolink connection to W8WKY-R and then look at hub.w8wky.org under the node 48496, does your Echolink node show up properly there?

Also, would you please upgrade your software to the latest versions? Those are all older and I want to make sure we're looking at a current bug.

Allan-N, 2024-10-13:
Did [Allmon3 PR#250](https://github.com/AllStarLink/Allmon3/pull/250) and [Allmon3 Release v1.3.5-1](https://github.com/AllStarLink/Allmon3/releases/tag/rel_t_1.3.5-1) address this issue?

jetski28, 2024-10-15:
I've just done an update/upgrade and its now looking good.
Allmon3 is now reporting the Echolink callsign.

 Package                         Version
        │   allmon3                         1.3.5-1.deb12
        │   asl3                            3.5.0-1.deb
        │   asl3-asterisk                   2:20.9.3+asl3-3.0.6-1.deb12
        │   asl3-asterisk-config            2:20.9.3+asl3-3.0.6-1.deb12
        │   asl3-asterisk-modules           2:20.9.3+asl3-3.0.6-1.deb12
        │   asl3-menu                       1.9-1.deb12

closed 2024-10-15 by Allan-N