AllStarLink/ASL3 #80: allstar-sys RSA auth fails

shaundolan1, 2024-07-06:
**Describe the bug**
The allstar.pub IAX2 RSA key is not installed, preventing RSA authentication via the allstar-sys context.

**To Reproduce**
1. Enter Asterisk CLI ```sudo asterisk -r```
2. Enable IAX2 debug messages ```iax2 set debug on```
3. Initiate node access connection via telephone portal
4. Observe warning from chan_iax2:
    ```help
    WARNING[21957]: chan_iax2.c:8261 authenticate_verify: Requested inkey 'allstar' for RSA authentication does not exist
    Tx-Frame Retry[000] -- OSeqno: 001 ISeqno: 002 Type: IAX     Subclass: REJECT
        Timestamp: 01087ms  SCall: 15399  DCall: 04086 [redacted]:4569
        CAUSE           : No authority found
        CAUSE CODE      : 50
    ```

**Expected behavior**
Call acceptance:
```help
Tx-Frame Retry[-01] -- OSeqno: 001 ISeqno: 002 Type: IAX     Subclass: ACK
    Timestamp: 00169ms  SCall: 14081  DCall: 10224 [redacted]:4569
Tx-Frame Retry[000] -- OSeqno: 001 ISeqno: 002 Type: IAX     Subclass: ACCEPT
    Timestamp: 00089ms  SCall: 14081  DCall: 10224 [redacted]:4569
```

**Software versions (listed in asl-menu, option 4)**
```text
OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.6.31+rpt-rpi-v8

Asterisk      : 20.8.1+asl3-3.0.0-1.deb12
ASL [app_rpt] : 3.0.0

Installed ASL packages :

Package               Version
====================  ==============================
allmon3               1.2.1-2
asl3                  3.1.1-1.deb
asl3-asterisk         2:20.8.1+asl3-3.0.0-1.deb12
asl3-asterisk-config  2:20.8.1+asl3-3.0.0-1.deb12
asl3-asterisk-module  2:20.8.1+asl3-3.0.0-1.deb12
asl3-menu             1.6-1.deb12
```

**Have you run a software update and rebooted?**
 - Yes

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - Raspberry Pi 3A+

**Additional context**
This appears to be due to the lack of the ```allstar.pub``` RSA key called for in the ```[allstar-sys]``` context of ```iax.conf```:
```ini
[allstar-sys]
type = user
context = allstar-sys
auth = rsa
inkeys = allstar
disallow = all
allow = ulaw
allow = adpcm
allow = gsm
```

**Workaround**
1. Verify the expected IAX2 keys location:
    ```shell
    $ sudo asterisk -rx 'core show settings' | grep IAX2
      IAX2 Keys directory:         /usr/share/asterisk/keys
    ```
2. Retrieve allstar.pub from the [archived ASL-Asterisk repo](https://github.com/AllStarLink/ASL-Asterisk/blob/d6a4b48d58288978bf2e5bf22e7e9cfcdea8b2f3/asterisk/keys/allstar.pub) and install it into the IAX2 keys directory with the [proper permissions](https://github.com/AllStarLink/ASL-Asterisk/blob/d6a4b48d58288978bf2e5bf22e7e9cfcdea8b2f3/asterisk/Makefile#L483):
    ```shell
    $ curl -sL https://github.com/AllStarLink/ASL- 
    Asterisk/raw/d6a4b48d58288978bf2e5bf22e7e9cfcdea8b2f3/asterisk/keys/allstar.pub | install -m 644 /dev/stdin 
    /usr/share/asterisk/keys -d -o asterisk:asterisk
    ```
3. Reload res_crypto in the Asterisk CLI with debug on:
    ```shell
    $ sudo asterisk -r
    allstar*CLI> core set debug 9 res_crypto.so
    allstar*CLI> module reload res_crypto.so
    ```
4. Verify that Asterisk reports loading the key upon res_crypto reload:
    ```help
        -- Loaded PUBLIC key 'allstar'
    ```
5. Turn debug off and exit the Asterisk CLI:
    ```shell
    allstar*CLI> core set debug 0 res_crypto.so
    allstar*CLI> exit
    ```

tsawyer, 2024-07-07:
Thank you for the bug find and specially for the fix.

jxmx, 2024-07-07:
https://github.com/AllStarLink/asl3-asterisk/issues/26

jxmx, 2024-07-07:
https://github.com/AllStarLink/ASL3/commit/3c4795899e971d5335776eaed380f6524171948d

jxmx, 2024-07-07:
This will appear in asl3_3.2.0-1

closed 2024-07-07 by jxmx

shaundolan1, 2024-07-08:
Amazing, thank you @jxmx!