AllStarLink/ASL3 #81: ASL3 ID is unreliable

kyle470, 2024-07-12:
**Describe the bug**
ID will often begin but not fully send ID audio. PTT is asserted but the ID is not always fully sent out. This occurs both with standard CWID as well as a custom ID audio file I have used for many years.

Manually sending the audio works reliably:
rpt playback 2212 filename

Just updated to latest ASL3 today and issue persists.

**Have you run a software update and rebooted?**
 - YES 

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Raspberry Pi4b

**Additional context**
Pi4b does have two RF CM108 nodes on it, but I have not noticed any correlation of this ID issue to having multiple RF nodes. This ID reliability issue happens on both RF nodes, even if one has been idle for hours.

KB4MDD, 2024-07-12:
You indicate that it starts but does not complete.  Is there any frequency to this problem?

Do you see any error or interesting messages in the cli?

kyle470, 2024-07-12:
I have not noticed a pattern necessarily. Sometimes it fails on the first
ID, other times the first ID after an idle period works fine. I have not
seen any interesting messages with debug level 3

kyle470, 2024-07-24:
Update: I have multiple Dell Wyse 3040 units running ASL3, and have not noticed the unreliable ID on these units. May only apply to the Rpi 4?

jxmx, 2024-08-10:
Move to app_rpt