AllStarLink/ASL3 #82: Connpgm and discpgm do not work

kyle470, 2024-07-12:
**Describe the bug**
connpgm and discpgm do not work. Currently have two boxes running ASL3, issue occurs on both. Understood from the Allstar Community site that this is an known issue and was expected to be resolved soon?

Updated to latest ASL3 today, issue persists.

**Have you run a software update and rebooted?**
 - YES 

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Raspberry Pi4b
  - Dell Wyse 3040

KB4MDD, 2024-07-12:
Do you see any error messages on the cli when a node connects or disconnects?

kyle470, 2024-07-12:
Normal messaging in CLI but no message indicating the connpgm or discpgm
command was sent.

I have tried
connpgm = touch /tmp/test

For example and the file was not touched.

I have placed the command in the template portion of rpt.conf and also
under each specific node definition,  does not appear to work.

Allan-N, 2024-07-12:
Can you to make sure that you have the latest updates.  You exec `sudo asl-show-version` to see what package versions have been installed.  For "asl3-asterisk", you want "2:20.8.1+asl3-3.0.2-1.deb12".  Prior versions had an issue with "connpgm".

kyle470, 2024-07-12:
********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.6.31+rpt-rpi-v8

Asterisk      : 20.8.1+asl3-3.0.2-1.deb12
ASL [app_rpt] : 3.0.1

Installed ASL packages :

  Package               Version
  ====================  ==============================
  allmon3               1.2.1-2
  asl3                  3.2.0-1.deb
  asl3-asterisk         2:20.8.1+asl3-3.0.2-1.deb12
  asl3-asterisk-config  2:20.8.1+asl3-3.0.2-1.deb12
  asl3-asterisk-module  2:20.8.1+asl3-3.0.2-1.deb12
  asl3-menu             1.6-1.deb12
  asl3-pi-appliance     1.7.4-1.deb12
  asl3-update-nodelist  1.4.0-1.deb12
  dahdi                 1:3.1.0-2
  dahdi-dkms            1:3.3.0-5+asl
  dahdi-linux           1:3.3.0-5+asl

Allan-N, 2024-07-12:
Here's what I just did :

1. I created a file, /usr/local/bin/asl-logger.sh, with the following :
```
#!/usr/bin/bash
echo $0 $*	>> /var/tmp/asl-logger-out.txt
```
2. I changed the permissions of the file with `sudo chown 755 /usr/local/bin/asl-logger.sh`

3. I edited /etc/asterisk/rpt.conf by adding the following two lines just after the commented out lines describing "connpgm" and "discpgm" :
```
connpgm = /usr/local/bin/asl-logger.sh 1
discpgm = /usr/local/bin/asl-logger.sh 0
```
4. Restart asterisk (`sudo astres.sh`)

5. connect to your node, disconect from your node, and check out /var/tmp/asl-logger-out.txt.  For me, I see :
```
/usr/local/bin/asl-logger.sh 1 491301 WA3WCO
/usr/local/bin/asl-logger.sh 0 491301 WA3WCO
```

kyle470, 2024-07-13:
Thanks for your assistance. I was running into some type of permissions issue apparently. I had been testing with files in a few other directories and not having any luck. I was also expecting to see an output on the CLI when the debug level was high enough.  My issue seems to be related to Asterisk running as user asterisk now, not root as in previous versions of ASL.

closed 2024-07-13 by kyle470