AllStarLink/ASL3 #83: Can't Edit crontab -e to enter the SkyWarnplus crontab entry

w8cpt, 2024-07-18:
While Installing Skywarnplus asl3 will not let me add the crontab entry???? i can't scroll down and paste the entry. It stops after the blue writing and wont go past the blue writing and I am in root??????

**To Reproduce**
Steps to reproduce the behavior:
1. root sudo su -
2. scroll down
3. stops at end of blue writing....

**Expected behavior**
scrool down to add another crontab entry

**Screenshots**
[image]

**Software versions (listed in asl-menu, option 4)**
 - asl3: lastest installed today
 - asl3-asterisk: latest installed today

**Have you run a software update and rebooted?**
 - YES or NO

**What is the platform - raspberry pi 3 a+

  - :Platform

**Additional context**
Add any other context about the problem here.

w8cpt, 2024-07-18:
I had to use this to get nano to work so I could add it...
 "EDITOR=nano crontab -e"

Allan-N, 2024-07-18:
It looks like you were using the "vi" editor.  I believe that, without the EDITOR env, you can use `update-alterniatives` to select your preferred editor (and you may need to do this both for your own login and for "root")

tsawyer, 2024-07-18:
Basic Linux instructions are better posted on Community. There is a wider audience there for such content. :)

closed 2024-07-18 by tsawyer