AllStarLink/ASL3 #84: Can't add a wifi network not in range

w8cpt, 2024-07-19:
**Describe the bug**
I followed the instructions to add a wifi network not yet in range and get an error.

**To Reproduce**
asl@node588840:~ $ nmcli conn add type wifi con-name ATTSq3EiVS ifname wlan0 ssid xxxxxxxxxxx nmcli conn modify ATTSq3EiVS wifi-sec.key-mgmt wpa-psk wifi-sec.psk xxxxxxxxxxxx
Error: invalid <setting>.<property> 'nmcli'.

**Expected behavior**
add a wifi network to use in the future at a different location

**Software versions (listed in asl-menu, option 4)**
 - asl3: [e.g. 3.1.1-1] Newest Installed today
 - asl3-asterisk: [e.g. 2:20.8.1+asl3-3.0.0-1] Newest installed today

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Pi 3 A+

**Additional context**
This is really difficult and adding and changing wifi networks should be easier....

encbar5, 2024-07-19:
The wiki needs to be updated. This is two commands, not one command. They must either be separated by a semicolon or entered one at a time.

`nmcli conn add type wifi con-name MYNETWORK ifname wlan0 ssid MYPASSWORD; nmcli conn modify MYNETWORK wifi-sec.key-mgmt wpa-psk wifi-sec.psk MYPASSWORD`

OR

`nmcli conn add type wifi con-name MYNETWORK ifname wlan0 ssid MYPASSWORD`
`nmcli conn modify MYNETWORK wifi-sec.key-mgmt wpa-psk wifi-sec.psk MYPASSWORD`

w8cpt, 2024-07-19:
OK thanks! I was pretty sure I copied and pasted correctly 🤣🤣🤣

closed 2024-07-19 by jxmx

w8cpt, 2024-07-21:
so i did this and am now at the location where I added the 2nd wifi for and it does not connect....

jxmx, 2024-07-21:
Need more info to help debug. Does `nmcli dev wifi list` show the network you're trying to connect to? Also are they any log entries in Cockpit's **Logs** panel?

w8cpt, 2024-07-21:
I'm pretty new to this I ended up changing my phone hotspot credentials to the credentials of where I set it up and ran this command while in range of the one I added when not in range and it connected. "nmcli device wifi connect MYNETWORK password MYPASSWORD" 

Before that I did the other two commands again twice and it probably has this current network in there 3 times now maybe 4. Is there somewhere that tells me the command to delete wifi networks so there's not 4 of the same? Im getting back on the laptop in a few if you want me to run the list. I ran it and saw the one I was connected to in green, a few in yellow that I assume are ones saved and then a bunch in blue that were available but not configured. I also need to shut down the node and see if it auto reconnects to this network now...it's frustrating that alot of people who do not know this kind of code are attempting this (like me) and causes alot of frustration for us learning and those helping us and I for one appreciate all of you helping so thanks!

w8cpt, 2024-07-21:
[image]
[image]
The one I'm currently connected to is showing twice green and yellow. That's the one I added before I was at this location and used the above code to add once I was here after I renamed my phones hotspot to access the cockpit....

w8cpt, 2024-07-21:
[image]
[image]
More log files in network. I'm gonna reboot it now and see if it connects now

w8cpt, 2024-07-21:
After the reboot it did connect again...here's the information after reboot...then mncli list after reboot
[image]
[image]

w8cpt, 2024-07-21:
I mean I supposed it's working but it shows the same one in yellow that's connected...not sure if it's ok to leave it like it is or if I vae to select the 4 of the same SSID I added which is also the one I'm currently connected to!

jxmx, 2024-07-21:
The "Chan 4" entry is the 2.4GHz radio and the "Chan 149" entry for the same SSID is the 5 GHz. All is as it should be assuming you're trying to connect to the SSID ending in 'EiVS'.

w8cpt, 2024-07-21:
OK thanks. It's working properly so I guess that's all that matters til it's not haha! Now to see if I can figure out how to create a dtmf command so *51 connects to the node I program it to...and *52 etc. I had that on my old node before ASL3 and it was handy....