AllStarLink/ASL3 #86: Autopatch issue - Cannot start tone on DAHDI/pseudo

kyle470, 2024-07-24:
Issue when trying to use Autopatch. Seems similar to the dreaded "cannot start dialtone" issue on previous versions:
rpt.conf entry was taken from an earlier version of ASL with the quiet=1 workaround:

[2024-07-24 10:01:18.087] WARNING[17064]: app_rpt/rpt_bridging.c:555 rpt_play_to                                                                             ne: Cannot start tone on DAHDI/pseudo-481952349
[2024-07-24 10:01:18.088] WARNING[17065][C-0000001d]: pbx.c:2928 pbx_extension_h                                                                             elper: No application 'Wait,1' for extension (icom-ctl, *01, 1)
[2024-07-24 10:01:18.098] WARNING[17064]: app_rpt.c:1352 rpt_call:  has no PBX?
[2024-07-24 10:01:18.118] WARNING[17064]: app_rpt/rpt_bridging.c:555 rpt_play_to                                                                             ne: Cannot start tone on DAHDI/pseudo-481952349

rpt.conf entry:
69=autopatchup,context=icom-ctl,noct=1,farenddisconnect=1,dialtime=90000,quiet=1

KB4MDD, 2024-07-24:
Autopatch appeared to be completely broken for me.  app_rpt froze when starting an autopatch call.  This was addressed in https://github.com/AllStarLink/app_rpt/pull/363 

Are you attempting to start a telephone call?

kyle470, 2024-07-24:
Not trying to make a phone call per se, using a SIP phone to control my Icom PCR1000 receiver. Worked fine on ASL 1 and Beta 2.

https://community.allstarlink.org/t/need-some-beta-testers-icom-civ-remote-base-control-from-the-dial-plan/19200/8

KB4MDD, 2024-07-24:
Hopefully the recent changes will be in the next beta so that you test to see if that solves the problem.

jxmx, 2024-08-10:
Moving to app_rpt