AllStarLink/ASL3 #89: monitor (record/archive) function not working as it used to

kenalker, 2024-08-01:
**Describe the bug**
I am using the “monitor” function to record repeater audio.  We have one node that mixes audio from three RTCMs plus anyone attaching to the node via Allstar.  Audio is recorded to a WAV file (date_and_time_stamp.WAV as filename) each time some keys up until they unkey (one file per key-up), as expected.  With the “previous” (very old) version of Allstar, every file contained audio, as would be expected.  With ASL3 I’m finding that most, but not all, files are 1K in length and contain no audio. The only files that are expected size and have audio appear to be from people who are connected to the node via Allstar.  Also, every now and then we get a file that is recorded from someone coming in from an RTCM that sounds like some kind of techo music (it’s very musically interesting, actually - maybe an incorrect CODEC or something), otherwise its just a 1K file with no audio content.

I reported this problem on the forum [here](https://community.allstarlink.org/t/monitor-record-function-not-working-properly-in-asl3/21510), but there isn't anything there that isn't in this report.  Also, Andriy_Grin posted the same issue [here](https://community.allstarlink.org/t/audio-archiving-still-the-same-in-asl3/21331/4) which includes a sample file of the "techno music" I referenced above, if you want to hear something fun and interesting but no intelligible audio.

**To Reproduce**
Steps to reproduce the behavior:
1) Add the archivedir command to your node stanza in rpt.conf.  Ensure directory where wave files will be written has proper permissions (ASL3 is not setting permissions to directory correctly when directory is created; I'll report this separately).  Here is a snippet of rpt.com:
[#####](node-main)
archivedir = /var/spool/asterisk/monitor
2) Key up your repeater and speak.
3) Play back the WAV file created in the /var/spool/asterisk/monitor/##### directory (where ##### is your node number)

**Expected behavior**
I expect to have a WAV (*.wav) file recorded that I can then play back just as would happen in the older version of ASL.

**Software versions (listed in asl-menu, option 4)**
OS: Debian GNU/Linux 12 (bookworm)
OS Kernel: 6.1.0-23-amd64
Asterisk: 20.8.1+asl3-3.0.2-1.deb12
ASL [app_rpt] : 3.0.1
Packages:
allmon3               1.2.1-2
asl3                  3.2.0-1.deb
asl3-asterisk         2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-config  2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-dev     2:20.8.1+asl3-3.0.2-1.deb12
asl3-asterisk-doc     2:20.8.1+asl3-3.0.2-1.deb12
Our previous system that worked flawlessly was many years old and I am unsure how to determine the ASL version but Asterisk was "Asterisk 1537 built by root @ repeater on a i686 running Linux on 2016-09-03 21:38:23 UT".

**Have you run a software update and rebooted?**
Yes.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Beelink SER5 Mini PC with AMD Ryzen 5 5560U 6 core (up to 4.0GHz) 16GB DDR4 RAM, 500GB NVMe M.2 SSD, RX Vega7

kenalker, 2024-08-01:
I just noticed this warning in the Asterisk console which may be helpful:
`[2024-07-31 20:17:39.417] WARNING[156961]: translate.c:408 framein: no samples for lintogsm`

kenalker, 2024-08-01:
I just bumped into [this] (https://www.asterisk.org/monitor-is-deprecated-farewell/) article that states, "The Monitor dialplan application and module in Asterisk ... has since been replaced by the MixMonitor dialplan application and module and will be removed as of Asterisk 2".  How does this apply to ASL3?  Should I be using mixmonitor.so instead of monitor.so?

jxmx, 2024-08-10:
Moving this to app_rpt.