AllStarLink/ASL3 #90: monitor (archive/record) doesn't work due to permissions problems with auto-created directories

kenalker, 2024-08-01:
**Describe the bug**
After invoking the monitor function of ASL3, errors and warnings are seen in the Asterisk console (asterisk -r) and no recordings are made.  In order to fix this, I changed permissions on the node number directory that was created by ASL3 when it tried to record.  In my case, I instructed ASL3 to record to /var/spool/asterisk/monitor (which I had already created, as I recall).  ASL3 created a directory under the monitor directory which is named 1900 (I made this up; ASL3 used my actual 5-digit Allstar node number).  The permissions on this file were "drw-------".  I changed permissions to "drwx------" (added the execute bit)  and the errors and warnings stopped popping up and ASL3 was able to write files to the node directory.  I do not know if my fix is really what should happen in the long-run, but it worked for me.  In the previous (old) version of ASL I was running, the execute bit was NOT set and the recordings were working just fine.  On my old system, the permissions on the node directory were "drw-------" which are the same as ASL3 is creating when creating a new node directory currently.  So, it seems the code that is generating the node directory hasn't changed, but for some reason ASL3 can not write to a directory with the these permissions.  Perhaps it has something to do with ASL3 running as the asterisk user (vs. root user as on the old system).

Additionally, I found that file.c had excellent data in the warning it was throwing that lead me to the problem, but app_rpt.c was throwing errors that were not as descriptive, so I'm going to suggest that someone enhance app_rpt.c to be more descriptive with this particular error (see "Additional context, below").

NOTE that I discovered this a month ago and I'm just now reporting it.  I took extensive notes while debugging but IT IS ENTIRELY POSSIBLE that I have reported something incorrectly or remembered something wrong, so, other than the main problem (permissions issue), please don't assume my report is 100% accurate.  I'm trying to give as much detail as possible, but the only way to verify everything I'm stating would be to do another fresh install from scratch and I don't have a system to do that on at the moment.

**To Reproduce**
Steps to reproduce the behavior:
1) Add the archivedir command to your node stanza in rpt.conf of a fresh ASL3 install.  Here is a snippet of rpt.com:
`[1900](node-main)`<br>`
archivedir = /var/spool/asterisk/monitor
`
2) Key up your repeater and speak.
3) Watch errors and warnings on the Asterisk console (asterisk -r).
4) Note that even though node directory under specified archivedir has been created, no files are being recorded.

**Expected behavior**
Files should be written without user having to modify permissions of node directory after it is created.

**Software versions (listed in asl-menu, option 4)**
I am running this now, but I was running something older a month ago (sorry I didn't record exactly what it was):
asl3                  3.2.0-1.deb
asl3-asterisk         2:20.8.1+asl3-3.0.2-1.deb12

**Have you run a software update and rebooted?**
No, too many changes to things since I wrote up my notes so wouldn't be a good thing to do now so my data is a little stale.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Beelink SER5 Mini PC with AMD Ryzen 5 5560U 6 core (up to 4.0GHz) 16GB DDR4 RAM, 500GB NVMe M.2 SSD, RX Vega7

**Additional context**
Here are the errors and warning that I saw when keying up while executing "asterisk -r":
`
[2024-06-22 05:47:37.378] ERROR[1184]: app_rpt.c:5772 rpt_master: Cannot open node log file 20240622054735,RXUNKEY,MAIN for write: Permission denied`<br>`
    -- Hungup 'DAHDI/pseudo-639788954'`<br>`
[2024-06-22 05:47:39.378] ERROR[1184]: app_rpt.c:5772 rpt_master: Cannot open node log file 20240622054737,TXUNKEY,MAIN for  write: Permission denied`<br>`
[2024-06-22 06:56:23.417] WARNING[1188]: file.c:1554 ast_writefile: Unable to open file /var/spool/asterisk/monitor/1900/20240622065623.WAV: Permission denied`<br>`
[2024-06-22 06:56:24.783] ERROR[1184]: app_rpt.c:5772 rpt_master: Cannot open node log file 20240622065623,TXKEY,MAIN for write: Permission denied`<br>`
    -- Hungup 'DAHDI/pseudo-22339593'`<br>`
[2024-06-22 06:56:28.783] ERROR[1184]: app_rpt.c:5772 rpt_master: Cannot open node log file 20240622065626,TXUNKEY,MAIN for write: Permission denied`

Please note that file.c reports EXACTLY the filename it can not write too, but app_rpt.c says it "Cannot open node log ..." which really didn't help me at all.  Where is this mysterious "node log" that it is trying to open?  Can someone make this spit out a more descriptive error similar to the warning being generated by file.c to help people in the future with debugging?

Allan-N, 2024-08-01:
> The permissions on this file were "drw-------". I changed permissions to "drwx------" (added the execute bit) and the errors and warnings stopped popping up and ASL3 was able to write files to the [...]

You must have started configuring/testing the "archivedir" functionality prior to the 3.0.2 update.  That update included a fix to correctly create the directories/files with the correct permissions.  But, that change only affects directories/files that are created, not any that already existed.  If your ".../monitor" directory already existed with the incorrect permissions then you would have needed to fix the permissions (as you did).

>  In the previous (old) version of ASL I was running, the execute bit was NOT set and the recordings were working just fine.

Previous (old) versions of ASL ran as the "root" users where filesystem permissions would have been an issue.

> Please note that file.c reports EXACTLY the filename it can not write too, but app_rpt.c says it "Cannot open node log ..." which really didn't help me at all. Where is this mysterious "node log" [...]

The change that went into 3.0.2 also updated the log message to include the full path.

For reference : https://github.com/AllStarLink/app_rpt/commit/dd4ff81b7dbb5392112ae76e65d0086597888328

Allan-N, 2024-08-01:
Note : you do want to make certain that the "archivedir" directory is writable by the "asterisk" user.
```
rpi:~ $ sudo ls -ld /var/spool /var/spool/asterisk /var/spool/asterisk/monitor
drwxr-xr-x 5 root     root     4096 Jul 12 10:33 /var/spool
drwxr-x--- 9 asterisk asterisk 4096 Aug  1 09:31 /var/spool/asterisk
drwx------ 3 asterisk asterisk 4096 Aug  1 09:31 /var/spool/asterisk/monitor
```

closed 2024-08-01 by Allan-N

Allan-N, 2024-08-01:
FYI, I just filed : https://github.com/AllStarLink/app_rpt/pull/379

kenalker, 2024-08-01:
> Note : you do want to make certain that the "archivedir" directory is writable by the "asterisk" user.
>
> ```
> rpi:~ $ sudo ls -ld /var/spool /var/spool/asterisk /var/spool/asterisk/monitor
> drwxr-xr-x [...]

What permissions should the actual node directories, themselves, have?  (The "#####" directories).
And what permissions should the .wav and .txt files within the nodes have?

Allan-N, 2024-08-01:
> Yes, I likely installed pre-3.0.2. Oh - wow; I was WONDERING if it was already fixed as I was seeing new files written with different permissions after finally upgrading, but I couldn't find any signs of anyone working on it in github.com/AllStarLink/ASL3 (I thought everything was under this project), so I spent the time writing above. I didn't realize there was also a github.com/AllStarLink/app_rpt. Darn! Well, it looks like some good came out of it anyway. Thanks so much for updating the logging too; very complete.

ASL3 is a big project comprised of numerous components.  The ASL3 repo has a few bits but is more of an umbrella for the project.  Any issue can be reported here with fixes being made in whatever repo has the code :-)

> What permissions should the actual node directories, themselves, have? (The "#####" directories).  And what permissions should the .wav and .txt files within the nodes have?

We create the "archivedir" directory with 0700 allowing "asterisk" to read/write the content.  The sub-directories are created with 0775 allowing "asterisk" to read/write, members of the "asterisk" group to read/write, and others to read.  Lastly, with the last fix that's "in the queue", the files in the leaf directories should be 0644.

Note:  the actual access to the sub-directories and leaf files are all dependent on whether YOU allow more access to the "archivedir" directory.  We start off with access being very limited and leave it up to you as to whether that access should be extended.