AllStarLink/ASL3 #91: Install fails on Debian 12

K1WIZ, 2024-08-06:
root@ASL3:~# wget https://github.com/AllStarLink/ASL3/releases/download/3.2.0-1/asl3_3.2.0-1.deb_all.deb
--2024-08-06 16:16:39--  https://github.com/AllStarLink/ASL3/releases/download/3.2.0-1/asl3_3.2.0-1.deb_all.deb
Resolving github.com (github.com)... 140.82.113.4
Connecting to github.com (github.com)|140.82.113.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/609928197/767a3573-f966-48f0-814b-1e0cbefc83fb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240806%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240806T201639Z&X-Amz-Expires=300&X-Amz-Signature=819ac4d9d2b7211dc8ad6c653cbbbb3b31291335aa48025c80bdf99eb0d83273&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=609928197&response-content-disposition=attachment%3B%20filename%3Dasl3_3.2.0-1.deb_all.deb&response-content-type=application%2Foctet-stream [following]
--2024-08-06 16:16:39--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/609928197/767a3573-f966-48f0-814b-1e0cbefc83fb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240806%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240806T201639Z&X-Amz-Expires=300&X-Amz-Signature=819ac4d9d2b7211dc8ad6c653cbbbb3b31291335aa48025c80bdf99eb0d83273&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=609928197&response-content-disposition=attachment%3B%20filename%3Dasl3_3.2.0-1.deb_all.deb&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.108.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9356 (9.1K) [application/octet-stream]
Saving to: ‘asl3_3.2.0-1.deb_all.deb’

asl3_3.2.0-1.deb_al 100%[===================>]   9.14K  --.-KB/s    in 0s      

2024-08-06 16:16:39 (29.0 MB/s) - ‘asl3_3.2.0-1.deb_all.deb’ saved [9356/9356]

root@ASL3:~# dpkg -i asl3_3.2.0-1.deb_all.deb 
Selecting previously unselected package asl3.
(Reading database ... 52347 files and directories currently installed.)
Preparing to unpack asl3_3.2.0-1.deb_all.deb ...
Unpacking asl3 (3.2.0-1.deb) ...
dpkg: dependency problems prevent configuration of asl3:
 asl3 depends on asl3-asterisk; however:
  Package asl3-asterisk is not installed.
 asl3 depends on asl3-menu; however:
  Package asl3-menu is not installed.
 asl3 depends on dahdi-linux; however:
  Package dahdi-linux is not installed.

dpkg: error processing package asl3 (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.11.2-2) ...
Errors were encountered while processing:
 asl3
 
 root@ASL3:~# apt install -f
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  asl3
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 41.0 kB disk space will be freed.
Do you want to continue? [Y/n] y
[master 1ba63f6] saving uncommitted changes in /etc prior to apt run
 2 files changed, 2 insertions(+), 2 deletions(-)
(Reading database ... 52364 files and directories currently installed.)
Removing asl3 (3.2.0-1.deb) ...
Processing triggers for man-db (2.11.2-2) ...
root@ASL3:~#

Allan-N, 2024-08-06:
Is there a reason you are not following the install instructions @ https://allstarlink.github.io/user-guide/install/

Allan-N, 2024-08-06:
The instructions to install ASL3 are available @ https://allstarlink.github.io/user-guide/install/

closed 2024-08-06 by Allan-N