AllStarLink/ASL3 #97: DTMF linking cut off for > 4 digit node numbers

dbehnke, 2024-08-23:
**Describe the bug**
When I attempt to do *3 + >4 digit node number, linking does not work.. digits after 4 get cut off

**To Reproduce**
Steps to reproduce the behavior:
1. monitor log /var/log/asterisk/messages
2. dial *3 ######
3. notice in log only *3 #### get read. last ## don't get acknowledged
4.  repeat the steps with a 4 digit node number, such as 2560, link works.
5. using a macro such as *51 and configuring macro to *803##### or similar does not work.
6. using rpt cmd [mynode] ilink 6 ##### works. 

**Expected behavior**
I can dial node numbers > 4 digits

**sample log**
This is me trying to connect to 594950 using dtmf, notice only 5949 gets picked up

```
[2024-08-23 11:42:13.480] DTMF[1252] channel.c: DTMF begin '*' received on SimpleUSB/550464
[2024-08-23 11:42:13.480] DTMF[1252] channel.c: DTMF begin passthrough '*' on SimpleUSB/550464
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end '*' received on SimpleUSB/550464, duration 447 ms
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end accepted with begin '*' on SimpleUSB/550464
[2024-08-23 11:42:13.928] DTMF[1252] channel.c: DTMF end passthrough '*' on SimpleUSB/550464
[2024-08-23 11:42:14.398] DTMF[1252] channel.c: DTMF begin '3' received on SimpleUSB/550464
[2024-08-23 11:42:14.398] DTMF[1252] channel.c: DTMF begin passthrough '3' on SimpleUSB/550464
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end '3' received on SimpleUSB/550464, duration 362 ms
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end accepted with begin '3' on SimpleUSB/550464
[2024-08-23 11:42:14.760] DTMF[1252] channel.c: DTMF end passthrough '3' on SimpleUSB/550464
[2024-08-23 11:42:15.123] DTMF[1252] channel.c: DTMF begin '5' received on SimpleUSB/550464
[2024-08-23 11:42:15.123] DTMF[1252] channel.c: DTMF begin passthrough '5' on SimpleUSB/550464
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end '5' received on SimpleUSB/550464, duration 298 ms
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end accepted with begin '5' on SimpleUSB/550464
[2024-08-23 11:42:15.422] DTMF[1252] channel.c: DTMF end passthrough '5' on SimpleUSB/550464
[2024-08-23 11:42:15.806] DTMF[1252] channel.c: DTMF begin '9' received on SimpleUSB/550464
[2024-08-23 11:42:15.806] DTMF[1252] channel.c: DTMF begin passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end '9' received on SimpleUSB/550464, duration 213 ms
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end accepted with begin '9' on SimpleUSB/550464
[2024-08-23 11:42:16.020] DTMF[1252] channel.c: DTMF end passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:16.445] DTMF[1252] channel.c: DTMF begin '4' received on SimpleUSB/550464
[2024-08-23 11:42:16.445] DTMF[1252] channel.c: DTMF begin passthrough '4' on SimpleUSB/550464
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end '4' received on SimpleUSB/550464, duration 255 ms
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end accepted with begin '4' on SimpleUSB/550464
[2024-08-23 11:42:16.701] DTMF[1252] channel.c: DTMF end passthrough '4' on SimpleUSB/550464
[2024-08-23 11:42:17.043] DTMF[1252] channel.c: DTMF begin '9' received on SimpleUSB/550464
[2024-08-23 11:42:17.043] DTMF[1252] channel.c: DTMF begin passthrough '9' on SimpleUSB/550464
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end '9' received on SimpleUSB/550464, duration 255 ms
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end accepted with begin '9' on SimpleUSB/550464
[2024-08-23 11:42:17.299] DTMF[1252] channel.c: DTMF end passthrough '9' on SimpleUSB/550464
```

Allan-N, 2024-08-23:
What version of ASL3 do you have installed?  If you are running a version prior to v3.0.4 then you are missing the fix for https://github.com/AllStarLink/app_rpt/pull/372.

Also, we recently had another reported issue, https://github.com/AllStarLink/app_rpt/issues/386, that turned out to be a network configuration issue.

From the node, do you have reasonable network connectivity?  Can you resolve/ping hostnames (e.g. `ping register.allstarlink.org`)?

dbehnke, 2024-08-23:
I'm checking the notes from the previous ticket there to see if this applys here.. in the meantime here's ping results..

```
ping register.allstarlink.org
PING register.allstarlink.org (162.248.92.131) 56(84) bytes of data.
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=1 ttl=56 time=30.2 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=2 ttl=56 time=25.6 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=3 ttl=56 time=26.2 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=4 ttl=56 time=25.9 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=5 ttl=56 time=22.0 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=6 ttl=56 time=32.6 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=7 ttl=56 time=34.9 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=8 ttl=56 time=28.4 ms
64 bytes from v-162-248-92-131.unman-vds.premium-chicago.nfoservers.com (162.248.92.131): icmp_seq=9 ttl=56 time=26.7 ms
```

Allan-N, 2024-08-23:
Great to hear that you have solved the problem.

Allan-N, 2024-08-23:
Dave,

Clearly, the simple "can we resolve "a" DNS name was not a good "quick" DNS test.  How did you determine that the issue was with Tailscale's DNS?  Is there a good test we could add (or at least document) that would help identify this issue?

dbehnke, 2024-08-23:
So for tailscale.. they have something called tailnet.. that allows you to do dns lookups to your other nodes.  and when you enable tailscale with `tailscale up` it will change your dns server to 100.100.100.100 which is tailscale's internal DNS.   I noticed this by looking at my /etc/resolv.conf that it was set to that.   If I temporily changed it to 8.8.8.8 in the /etc/resolv.conf, I was then able to connect to allstar nodes.   So what I think happens when you have it set to 100.100.100.100 it tries to then bounce to your internal address .. in my case 192.168.0.53 is my dns locally.  and it can't do that from its point of view in the cloud.   One thing I didn't try (in the screenshot below) is to flip the switch to override local DNS... i'm wondering if that would also work.

[image]

jxmx, 2024-08-25:
This is likely related to a know DNS services defect on the infra side. Try not using the intercepted DNS.

dbehnke, 2024-08-25:
Yeah I've kinda given up for now with this .. even with tailscale on and disabling intercepting DNS (using 8.8.8.8) it still has an issue.  The only way i was able to work around the issue for now is to use FILE mode in rpt.conf (turning off DNS lookup) or turn off tailscale and then DNS works.

I use tailscale as a vpn between my nodes and as an exit node for when i'm on cellular to appear with the same ip. (ATT likes to change IPs unfortunately). Not sure what's changed between first release and 3.0.4 but something broke using tailscale for me.

KB4MDD, 2024-08-26:
@dbehnke Would you be willing to setup your system with the failing configuration for me to get some debugging information.  We need to at least understand why this causes the failure.

If you are willing have some questions and steps.