AllStarLink/ASL3 #113: DTMF via IAX VOIP connection fails

wspivak, 2024-12-19:
Using HOIP on an Asterisk 20.10.0+asl3-3.1.0-1.deb12, DTMF fails after connection to the private node.

The HOIP softphone connects in monitor mode, but I am unable to get node to decode *99.  Gescio Alpuro, WH6AV of HOIP tried to get it to work on my node without success.  Before latest update, it WAS operational.

Currently: DTMF works via CLI and/or RF connected to node.

Tried setting jitterbuffer to "NO"  (via Jason N8EI) and it refuses to maintain a connection to HOIP.

Allan-N, 2024-12-19:
When you say "Before latest update, it WAS operational.", what update are you referring to?  Are you saying that you have been regularly updating your node and a recent ASL3 update/upgrade changed the behavior without your having made any configuration changes?

Can you attach the `rpt.conf` and `extensions.conf` files from your node?

... and, if not obvious in `rpt.conf`, add some notes to this issue noting the node(s) in play (e.g. what nodes are connected, what node you are entering *99, etc)

... and, yes, you should remove any passwords?

Note: you may need to add a ".txt" extension to the filenames to keep GitHub happy.

wspivak, 2024-12-19:
Right now Jessie is researching, so let me wait till he’s done and report back if he found a reason from his side…

 

See answers below

 

I update the Pi daily, and ASL did a update a week or two ago???, well I updated.  I changed nothing with the HOIP node (1500) or IAX/Extensions from the time I installed.  Since that time, I’ve updated the Extensions commands per Jason (Jessie agreed) and I, today as I said changed jitterbuffer.  Other than that, no changes.

wspivak, 2024-12-19:
I guess he wasn't successful, so the files are here that were requested.

[iax.conf.txt](https://github.com/user-attachments/files/18197301/iax.conf.txt)
[extensions.conf.txt](https://github.com/user-attachments/files/18197302/extensions.conf.txt)
[rpt.conf.txt](https://github.com/user-attachments/files/18197303/rpt.conf.txt)