AllStarLink/ASL3 #114: Ensure that .conf files are not stepped on during 'apt upgrade'

Allan-N, 2024-12-21:
We would like to make changes to the default/template configuration (.conf) files.  But, we also want to ensure that any of our software updates do not, inadvertently, step on an existing configuration.  Working configurations need to keep working.

For now, "I" think we need to just leave any existing configuration file as-is.  While not required at this time, having a "hook" that would allow us to exec a script/program that would help migrate/modify an existing configuration file would be a plus.

InterLinked1, 2024-12-21:
Don't Debian packages already have provisions for managing this? When upgrading, I often get a message "the package has a newer config, what would you like to do?" with options to either keep my existing file or use the new one.

jxmx, 2024-12-21:
Yes it does. There's a number of people who keep saying "Y" to random files to replace them breaking their configuration. There's an associated pull I just make that will unconditionally answer "N" (keep) to all package installs.

jxmx, 2025-01-01:
Fixed by #115

closed 2025-01-01 by jxmx