AllStarLink/ASL3 #120: ASL3 IAX Default Port Change - Nodes 10xx wont work

m0llc, 2025-01-21:
**Describe the bug**
When I change the default IAX port from 4569 to say 4570, Extensions 1001 - 1002 etc will not connect to 127.0.0.1. I recently setup a zello bridge, 1001 would only dial if IAX port was set to default 4569. No Firewalls, Fresh Debian 12 installs and happend on all of my ASL3 installs.
Extensions conf set correct etc. 

**Software versions (listed in asl-menu, option 4)**
ASL3 Latest as of today 21/01/25

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :pi4 8gb and VPS Debian 12 Cloud hosted

jxmx, 2025-01-21:
Can you please clarify your issue? When you say "1001 would only dial", what are you talking about? This sounds like the Zello bridge can't use a non-standard port.  There is no issue with nodes < 1999 using a non-standard port.

m0llc, 2025-01-22:
Sorry. Any extension I create using USRP 1001 1002 1003 etc that is sent to 127.0.0.1 never works with the IAX port not set to standatd 4569. Even data Tests to the USRP ports result in the test freezing. Soon as I changed the port back to 4569 it all worked no issues. Anything I did and someone else who had a go could not get any local USRP 127.0.0.1 extensions to work. Hope this explains. Didnt have a issue with ASL 2.0 Beta.

jxmx, 2025-01-22:
I still don't understand what you're trying to do. By "USRP" I assume you're referring to a USRP hardware device that is elsewhere on the network? So what's the "USRP 1001" stuff and 127.0.0.1?

m0llc, 2025-01-22:
The issue is that any local USRP extensions I create (e.g., 1001, 1002, 1003) that are supposed to connect via 127.0.0.1 will not work if the IAX port is changed from the default 4569 to anything else (e.g., 4570). This includes basic data tests to the USRP ports, which freeze when using a non-default port. As soon as the IAX port is reverted back to 4569, everything works perfectly.

This behavior suggests that local USRP extensions or connections to 127.0.0.1 are somehow locked to the default IAX port (4569) and cannot function correctly when the port is changed.

This was not an issue with ASL 2.0 Beta, where I was able to use non-default IAX ports without any problems. It seems like there might be a bug or limitation in ASL3 related to handling local USRP extensions when using non-standard IAX ports.

I bridge 2 extensions over the USRP Protocal to various things and I use 1001 and 1002. If I change the port from 4569 to something else in IAX.Conf and if the extension is set to someting like :

1001   = radio@127.0.0.1/1001,NONE
1002   = radio@127.0.0.1/1002,NONE

[1001](node-main)
duplex = 0
hangtime = 100
rxchannel = USRP/127.0.0.1:7070:7071

[1002](node-main)
duplex = 0
hangtime = 100
rxchannel = USRP/192.168.2.231:32010:34010

In rpt.conf
The Channel 1001 will not work when the default IAX port is changed.

You could replicate this by setting up a new instance
Changing port from 4569 to say 4570 in Iax.conf
Enable USRP Channel in Moules
Then trying to connect asterisk to 1001.. 
It doenst work
BUT with default port of 4569 in iax.conf it does.

KB4MDD, 2025-01-22:
Can you provide the command you are using to "connect asterisk to 1001"?

m0llc, 2025-01-22:
Im using Allmon3 - Connect 1001

KB4MDD, 2025-01-22:
Let me go another route.  Can you ssh into the device and run `asterisk -r` .  Since this is not working, we should see some output on the cli interface that would help.

Allan-N, 2025-01-23:
Q? how are you changing the IAX port?
- Are you using the ASL3 menu / "Node Setup"?
- or are you directly editing the `/etc/asterisk/iax.conf` file?

If the latter, are you also updating the `[nodes]` section of `rpt.conf`?

m0llc, 2025-01-23:
im manually changing it via IAX.conf

Allan-N, 2025-01-23:
The IAX port is set in `iax.conf`.  If you change the port then the references in the `[nodes]` section of `rpt.conf` must be updated to match.

In your `rpt.conf` you would use the following for IAX port 4589 : 
```
1001 = radio@127.0.0.1/1001,NONE
```
For IAX port 4570 you would use :
```
1001 = radio@127.0.0.1:4568/1001,NONE
```

Note: the ASL3 Menu takes care of this for you.

closed 2025-01-23 by Allan-N