AllStarLink/ASL3 #124: cop,48 produces choppy and broken telemetry

AI3I, 2025-02-01:
**Describe the bug**
cop,48 in app_rpt does not produce consistent audio fidelity with tone playback, either with single tones, mixed tones/tone sequences, or pre-programmed tones, i.e. dual-tone multi-frequency (DTMF) sequences.  The bug seems to manifest as though it has a buffering issue; it may, for example, produce one or two DTMF digits from a named sequence and not produce the rest, or skip through sequences, or hold on and play some tone sequences endlessly.

**To Reproduce**
Two examples that 'stress' cop,48 behavior:
`node504381*CLI> rpt cmd 504381 cop 48 !830+827/450,!0/250,!659+656/450,!0/250,!739+736/450,!0/250,!493+490/450,!0/750,!493+490/450,!0/250,!739+736/450,!0/250,!830+827/45 0,!0/250,!659+656/450`  ...should emulate a Computer Automated Technology CAT-1000 grandfather clock feature, but produces clipped audio and completes one or two tone sequences before playing what remains in its buffer endlessly.  Results vary.

`node504381*CLI> rpt cmd 504381 cop 48 6,5,4,5,6,6,6,5,5,5,6,#,#`  ...should whimsically play 'Mary Had a Little Lamb' in DTMF, but produces 6, 5, may skip other digits, perhaps #, and stops.  Results vary.

Playing multiple sequences may even return chan_simpleusb errors, with Asterisk abruptly restarting:
`[2025-01-31 21:19:29.073] WARNING[1049]: chan_simpleusb.c:2073 simpleusb_read: Possibly stuck USB read channel. [504381]
[2025-01-31 21:19:29.135] WARNING[1049]: chan_simpleusb.c:2084 simpleusb_read: USB read channel [504381] was not stuck.`

Additional logging during observed behavior (from /var/log/asterisk/messages.log):
`[2025-01-31 20:36:07.311] Asterisk 20.11.0+asl3-3.2.0-2.deb12 built by builder @ allstarlink.org on a aarch64 running Linux on 2024-12-20 20:30:41 UTC
[2025-01-31 20:36:07.404] NOTICE[883] loader.c: 70 modules will be loaded.
[2025-01-31 20:36:07.930] NOTICE[883] cdr.c: CDR simple logging enabled.
[2025-01-31 20:36:07.937] NOTICE[883] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2025-01-31 20:36:08.003] NOTICE[883] indications.c: Default country for indication tones: us
[2025-01-31 20:36:08.003] NOTICE[883] indications.c: Setting default indication country to 'us'
[2025-01-31 20:36:08.223] NOTICE[883] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2025-01-31 20:36:08.261] WARNING[883] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2025-01-31 20:36:08.261] WARNING[883] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2025-01-31 20:36:08.261] NOTICE[883] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 23.
[2025-01-31 20:36:08.261] NOTICE[883] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 31.
[2025-01-31 20:36:08.261] NOTICE[883] chan_dahdi.c: Ignoring any changes to 'hassip' (on reload) at line 35.
[2025-01-31 20:36:08.261] NOTICE[883] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 39.
[2025-01-31 20:36:08.261] NOTICE[883] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 47.
[2025-01-31 20:36:08.261] WARNING[883] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2025-01-31 20:36:08.261] WARNING[883] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2025-01-31 20:36:08.780] WARNING[883] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2025-01-31 20:36:09.096] NOTICE[1034] app_rpt.c: Normal Repeater Init  504381    
[2025-01-31 20:36:09.105] NOTICE[1049] chan_simpleusb.c: Channel 504381: Set option TONE VERIFY, mode: OFF(0).
[2025-01-31 20:36:09.180] WARNING[1049] chan_simpleusb.c: Possibly stuck USB read channel. [504381]
[2025-01-31 20:36:09.206] WARNING[1049] chan_simpleusb.c: USB read channel [504381] was not stuck.
`

**Expected behavior**
cop,48 should playback with fidelity, no choppiness, and be able to handle all buffered tone sequences without clipping, halting, and observing internal timings for pre-defined sequences (either DTMF) or timings declared in explicit timings.  It should be noted that precision values in previous versions of app_rpt with decimal values were played back; whether rounded to the nearest cycle or generated with precision remains to be seen.

**Screenshots**
No screenshots are available for this report.

**Software versions (listed in asl-menu, option 4)**
********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.6.62+rpt-rpi-v8

Asterisk      : 20.11.0+asl3-3.2.0-2.deb12
ASL [app_rpt] : 3.2.0

Installed ASL packages :

  Package                         Version
  ==============================  ==============================
  allmon3                         1.4.2-1.deb12
  asl3                            3.6.0-1.deb
  asl3-asterisk                   2:20.11.0+asl3-3.2.0-2.deb12
  asl3-asterisk-config            2:20.11.0+asl3-3.2.0-2.deb12
  asl3-asterisk-modules           2:20.11.0+asl3-3.2.0-2.deb12
  asl3-menu                       1.11-1.deb12
  asl3-pi-appliance               1.9.0-5.deb12
  asl3-update-nodelist            1.5.1-1.deb12
  cockpit                         287.1-0+deb12u3
  cockpit-bridge                  287.1-0+deb12u3
  cockpit-networkmanager          287.1-0+deb12u3
  cockpit-packagekit              287.1-0+deb12u3
  cockpit-sosreport               287.1-0+deb12u3
  cockpit-storaged                287.1-0+deb12u3
  cockpit-system                  287.1-0+deb12u3
  cockpit-wifimanager             1.1.1-1.deb12
  cockpit-ws                      287.1-0+deb12u3
  dahdi                           1:3.1.0-2
  dahdi-dkms                      1:3.4.0-4+asl
  dahdi-linux                     1:3.4.0-4+asl

**Have you run a software update and rebooted?**
YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
root@node504381:~# cat /proc/cpuinfo | grep Model 
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

**Additional context**
It is not readily known whether this is fully reproducible on X86_64 platforms, but should be tested.  It is also not known whether this is specific to or extends beyond chan_simpleusb.

AI3I, 2025-02-01:
Observed as early as July 2024 from ASL forums:  [https://community.allstarlink.org/t/am-i-the-only-seeing-cop-48-weirdness-in-asl3/21299]

Allan-N, 2025-02-01:
The July 2024 community reference suggests a regression between ASL2 and ASL3

jxmx, 2025-02-01:
Moving to app_rpt