AllStarLink/ASL3 #136: Sounds do not play running allstar3-arm64-3.0.17.img.xz

jimrochny, 2025-04-01:
ASL3 is using the wrong default directory to play sound files, and not searching the three locations.
Executing a command that expects an audio response, produces nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run sudo asterisk -rvvv to enter CLI
2. rpt fun <nodenumber> <link command>
3. Watch result and view errors
4. ASL3 is attempting to play from what appears to the be ASL2 legacy directory /var/lib/asterisk/sounds/custom instead of /usr/share/asterisk/sounds/en/ which "should" be the default?
5. Edit /etc/asterisk/asterisk.conf and change sounds_search_custom_dir = yes to "= no".  Save and Restart asterisk.  Temp fix.
6. Repeat step 2 and sounds starting working.

**Expected behavior**
Per the manual, ASL3 is supposed to search the /var/... directory first then /usr/... but it never gets that far.  I'm not sure why if /usr/... is the new default, it doesn't check there first, I am unfamiliar how to change the search order either.

**Software versions (listed in asl-menu, option 4)**
  OS            : Debian GNU/Linux 12 (bookworm)
                   │ OS Kernel     : 6.12.20+rpt-rpi-v8
                   │
                   │ Asterisk      : 22.2.0+asl3-3.3.0-6.deb12
                   │ ASL [app_rpt] : 3.3.0
                   │
                   │ Installed ASL packages :
                   │
                   │   Package                         Version
                   │   ==============================  ==============================
                   │   allmon3                         1.4.2-1.deb12
                   │   asl3                            3.7.1-1.deb
                   │   asl3-asterisk                   2:22.2.0+asl3-3.3.0-6.deb12
                   │   asl3-asterisk-config            2:22.2.0+asl3-3.3.0-6.deb12
                   │   asl3-asterisk-modules           2:22.2.0+asl3-3.3.0-6.deb12
                   │   asl3-menu                       1.12-1.deb12

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Raspberry Pi3 B+

Allan-N, 2025-04-01:
>5. Edit /etc/asterisk/asterisk.conf and change sounds_search_custom_dir = yes to "= no". Save and Restart asterisk. Temp fix.

The issue with the WARNING messages has been addressed and has been included in our next released (currently in "beta" testing).  For more info, see [Asterisk PR #1172](https://github.com/asterisk/asterisk/pull/1172), [asl3-asterisk PR #46](https://github.com/AllStarLink/asl3-asterisk/pull/46)

>Per the manual, ASL3 is supposed to search the /var/... directory first then /usr/... but it never gets that far. I'm not sure why if /usr/... is the new default, it doesn't check there first, I am unfamiliar how to change the search order either.

What "manual" are you referring to ?

As for where ASL3 searches for files, Asterisk uses its "Data directory" (asterisk.conf, `astdatadir => /usr/share/asterisk`) as the root for searches.  If `sounds_search_custom_dir = yes` then the code first looks in `/usr/share/asterisk/sounds/custom`.  Then, the code looks to `/usr/share/asterisk/sounds`.  One added wrinkle is that `/usr/share/asterisk/sounds/custom` is actually a symlink to `/usr/local/share/asterisk/sounds`.  In short, there are 1 (or 2) sound directories in use:
```
/usr/local/share/asterisk/sounds/...   (used when "sounds_search_custom_dir = yes")
/usr/share/asterisk/sounds/...
```

jimrochny, 2025-04-02:
"Per the manual, ASL3 is supposed to search the /var/... directory first then /usr/... but it never gets that far"
What "manual" are you referring to ?

I stand corrected, the other way around.  Okay on fix coming out with next release.  Temp =no "fix" in place for now for 3 different sites I'm rolling out.  Thanks for all the work in straightening out years of aging ASL code.  Everything is so much easier to sort through and use now.

closed 2025-04-02 by jimrochny

Allan-N, 2025-04-02:
Thank you for your support!