AllStarLink/ASL3 #137: ASL3 Debian12 glitches in USB Rx audio

davidgsd, 2025-04-09:
I have a node that had and still has some intermittent glitches in the USB rx audio (from URI ADC). This node had not been updated in a few months so I updated it to latest ASL3.3 and the issue was still there. I then tried updating to latest devel and that doesn't fix it either. sudo asterisk -V shows: Asterisk 22.2.0+asl3-3.4.2-1.deb12

I have another node with the same URI and everything is fine on that, it has Asterisk 20.11.0+asl3-3.2.0-2.deb12.

Here is what the glitches look like:
glitches
[glitches1920×1131 175 KB](https://community.allstarlink.org/uploads/default/original/2X/5/5c3afde68a2ab8affb409f7bbbd89aba756ff6d9.jpeg)

(from parrot wav file recorded by the node in /tmp/)
ie. there are anywhere from 5 ~15 zero's in the PCM data, but only happens intermittently, maybe once every 10 to 20 seconds of audio, with maybe 8 or 9 out of 10 20mS USB frames then having a sequence of zero sample values (presumably at frame boundaries), resulting in some choppiness/distortion in the audio.

Please LMK of any suggestions on how to further debug this issue.

**To Reproduce**
Set up a node and do a parrot test and listen for distortion / "choppiness" in the played back audio. My node has duplex=3 and is using SimpleUSB channel driver.

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Dell Wyse 3040 MiniPC

mkmer, 2025-04-09:
I believe this is related to this PR https://github.com/AllStarLink/app_rpt/pull/598
Even with the kernel set to `options snd_usb_audio lowlatency=0` we are under running the buffer, just not "stopping" the audio.

Should be in next month's release, early testing is appreciated!

davidgsd, 2025-04-11:
When I updated to latest devel a few days ago (to 22.2.0+asl3-3.4.2-1.deb12) shouldn't that have already contained this PR? Or was that PR only in beta but not devel? Also that PR seems to have changed only sending of Tx audio frames (to the USB device)? whereas the issue I'm seeing is with audio from the USB device.  Thanks

mkmer, 2025-04-11:
Ok - I misunderstood the direction you were referring too. Sorry about that.
The RX side is driven by the device (I think it's polled rather than interrupt driven).  For "inter frame" zeros it think we would have to be under run by the device.  I'll do some digging.
You say Asterisk 20.11.0+asl3-3.2.0-2.deb12 appears to work fine, asl3-3.3.x was broken as well?

Allan-N, 2025-04-11:
>I just updated this node to 22.2.0+asl3-3.4.3-1.deb12 (Beta)

FYI : 3.4.3 corrected some EchoLink issues

mkmer, 2025-04-11:
Not seeing any errors or warnings (like these)?
"Stuck USB read channel [%s], un-sticking it!\n"
"Possibly stuck USB read channel. 
"USB read channel [%s] was not stuck.\n",

I suppose the next step is to capture the raw incoming audio.  In the CLI, `susb tune rxcap` and `susb tune txcap`

jxmx, 2025-04-12:
Moving this to app_rpt and dispositioning accordingly