AllStarLink/ASL3 #138: How to install a All-Star hub in the cloud that works with EchoLink and All Star, using RepeaterPhone or any radio less connection?

KQ4UPK, 2025-04-09:
I’m using vulture. And I am a noob at this, my buddy in South Africa ZS6ZYM Is blind and he had a friend do exactly what I am trying to do for him. Can anybody explain it to me step-by-step or give me a good bit of guidance? Need to do I know how to install all Star link 3 and Allmon And all the updates I’m pretty good with Linux. 

They simply want a hub that I can dial up to using my iPhone app repeater phone and/or others. Can dial up to Almost like a conference server, I guess. And also if anybody has repeaters that want to connect to the hub That would be fine if I knew how to do that on my end like I said I am quite a noob to this type of install and configuration so please anyone who can walk me through this please help a fellow HAM out. 

Ty from KQ4UPK

Allan-N, 2025-04-10:
Hopefully, you have looked over the [AllStarLink Manual](https://allstarlink.github.io).

The manual includes a section, [Debian 12 Install](https://allstarlink.github.io/user-guide/install), that covers the basics of installing ASL3 to a new OS on a PC or VM.

The manual also includes a section, [Allmon3](https://allstarlink.github.io/allmon3/install), that covers installation and setup of Allmon3.

What we don't cover are the specifics of installing and setting up the OS, setting up the network (typically, to support incoming calls, you want to allow network traffic from UDP port 4569), and installing/setting up any other applications you feel are needed on your node.

If you need help, I will direct you to the [AllStarLink Community](https://allstarlink.github.io/user-guide/support)

closed 2025-04-10 by Allan-N

KQ4UPK, 2025-04-10:
Please I do need help with the UDP Port. I need to know exactly how to do
it. With My VM and how to setup my Xfinity Gateway (router) I would love
for you to direct me on the path to successfully getting my Cloud server
installed and working as you know I would like it to.

Ty very much
Chris
KQ4UPK

Allan-N, 2025-04-10:
>Please I do need help with the UDP Port. I need to know exactly how to do it.

I can't help you here.  For home networks you would typically need to use your routers admin interface to setup port forwarding. I have no insights into how you would configure a Cloud VM.

KQ4UPK, 2025-04-10:
Ty Allan, I just got finished reading the whole AllStar asl3 manual. I wish
my friend in South Africa call sign ZS6ZYM, His name is Jan. He is a blind
Ham he has a hub node running in the cloud that will accept a connection
from any app android DV Switch and Repeater phone on iPhone using AllStar
or EchoLink I would love to ask him if it would be possible for his friend
who did it for him to do it for me.

And do a complete screen record from
Beginning to End. To be able to share this process with the rest of us who
would love to skip the whole process of using a raspberry pi. (With these
apps DV switch on Android phones and RepeaterPhone on IPhone) makes it
radio less next level Tech.
As a friend of mine put it.

I’ve looked all over YouTube and the only thing that really gives me
step-by-step instructions on how to do. This is ChatGPT, but I no longer
have access to it due to outrageous subscription fees.

I hope they get a hold of Jan tomorrow afternoon because he is 7hrs ahead
of us on EST. I spoke with him today about it but he just kept saying in so
many ways. To my understanding he does not have any idea how to do it.

His Node is 41042 and it’s always so stable and reliable running 24/7 it’s
quite amazing.

Most people who have Nodes are aware of the East Coast Reflector.

Well I think it’s nice but because of how many raspberry pi’s are running
in the server room and the continuous packet loss. It’s becoming very
annoying. Also
Lately the only time I ever hear anyone talking is during the nets.

Well then Ty again and I hope to
Have this Cloud Hub Node Running asap.

Best outcome would be to have Jan’s friend do it and record the whole
thing.

There are many ways that he and I can collaborate on this project to get a
real easy to understand step by step process to make this cloud Hub server
(AllStar node)

It will be labeled KQ4UPK (node#64921)