AllStarLink/ASL3 #141: rpt reload crashes ASL3

kyle470, 2025-04-13:
**Describe the bug**
I have ASL3 installed (and updated) on several Dell Wyse 3040. Anytime I issue:

rpt restart

from the CLI, Asterisk crashes and must be restarted. 

3040-4*CLI> rpt restart
[2025-04-13 17:00:43.072] WARNING[1644086]: app_rpt.c:5789 rpt_master: rpt_thread restarted on node 2216
[2025-04-13 17:00:43.072] WARNING[1644086]: app_rpt.c:5789 rpt_master: rpt_thread restarted on node 2217
[2025-04-13 17:00:43.081] NOTICE[1871069]: chan_usbradio.c:2560 usbradio_setoption: Channel 2217: Set option TONE VERIFY, mode: OFF(0).
[2025-04-13 17:00:43.081] NOTICE[1871068]: chan_usbradio.c:2560 usbradio_setoption: Channel 2216: Set option TONE VERIFY, mode: OFF(0).
[2025-04-13 17:00:43.107] ERROR[1871070]: chan_usbradio.c:1597 setformat: Channel 2216: Unable to open DSP device 1: Device or resource busy.
[2025-04-13 17:00:43.108] ERROR[1871071]: chan_usbradio.c:1597 setformat: Channel 2217: Unable to open DSP device 0: Device or resource busy.
[2025-04-13 17:00:46.207] ERROR[1871068]: chan_usbradio.c:1962 usbradio_read: Channel 2216: HID process has died or is not responding.
[2025-04-13 17:00:46.207] ERROR[1871069]: chan_usbradio.c:1962 usbradio_read: Channel 2217: HID process has died or is not responding.
3040-4*CLI>

A clear and concise description of what the bug is.

On older versions, I would use rpt reload, which worked fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software versions (listed in asl-menu, option 4)**
 - asl3: [e.g. 3.1.1-1]
 - asl3-asterisk: [e.g. 2:20.8.1+asl3-3.0.0-1]

**Have you run a software update and rebooted?**
 - YES or NO

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Platform

**Additional context**
Add any other context about the problem here.

mkmer, 2025-04-14:
https://allstarlink.github.io/adv-topics/incompatibles/

kyle470, 2025-04-14:
Hello. Thanks for that info, although my ASL3 installs were all done from the official ASL3 image, so this rule exists on each box already.

/etc/udev/rules.d/90-asl3.rules
SUBSYSTEM=="usb", ATTRS{idVendor}=="0d8c", GROUP="plugdev", TAG+="uaccess"

Rebooting, and/or removing and re-inserting USB device did not help either.

mkmer, 2025-04-14:
It sounds like either the user asterisk doesn't have permission to access the device OR something else has exclusive access.  
Can you update the issue to include the versions, platform, etc 
Are you on Debian 12?  Does it only crash on reload?

mkmer, 2025-04-15:
https://github.com/AllStarLink/app_rpt/pull/631

mkmer, 2025-04-15:
This one is "fixed" - will be in our next release (Mayish).  The best thing to do until then is restart asterisk at the command line..
something like `sudo systemctl restart asterisk.service` rather than `rpt restart` until the release.

kyle470, 2025-04-15:
Many thanks

InterLinked1, 2025-04-17:
> This one is "fixed" - will be in our next release (Mayish). The best thing to do until then is restart asterisk at the command line.. something like `sudo systemctl restart asterisk.service` rather than `rpt restart` until the release.

What happens with `module refresh app_rpt`? Any difference?

kyle470, 2025-04-17:
This had a very similar effect to rpt restart: Both required me to restart
Asterisk:

3040-4*CLI> module refresh app_rpt.so
Unloaded and loaded app_rpt.so
[2025-04-17 10:57:47.855] WARNING[1956044]: app_rpt.c:5879 rpt_master:
Failed to join 2216 thread: Success
[2025-04-17 10:57:47.855] WARNING[1956044]: app_rpt.c:5879 rpt_master:
Failed to join 2217 thread: Success
[2025-04-17 10:57:47.880] WARNING[1956049]: app_rpt/rpt_bridging.c:134
rpt_hangup: No rxchan channel to hang up
[2025-04-17 10:57:47.881] WARNING[1956051]: app_rpt/rpt_bridging.c:134
rpt_hangup: No rxchan channel to hang up
[2025-04-17 10:57:48.062] NOTICE[1973631]: app_rpt.c:5730 rpt_master:
Normal Repeater Init  2216
[2025-04-17 10:57:48.063] NOTICE[1973632]: chan_usbradio.c:2661
usbradio_request: Channel 2216: Already have a call (chan 0x7f2e04001760)
on the usb channel
[2025-04-17 10:57:48.063] ERROR[1973632]: app_rpt/rpt_bridging.c:231
__rpt_request: Failed to request Radio/2216
[2025-04-17 10:57:48.063] WARNING[1973632]: app_rpt.c:2904 disable_rpt:
Disabled broken repeater 2216
[2025-04-17 10:57:48.065] NOTICE[1973631]: app_rpt.c:5730 rpt_master:
Normal Repeater Init  2217
[2025-04-17 10:57:48.070] NOTICE[1973633]: chan_usbradio.c:2560
usbradio_setoption: Channel 2217: Set option TONE VERIFY, mode: OFF(0).
[2025-04-17 10:57:48.118] WARNING[1973633]: chan_usbradio.c:2040
usbradio_read: Possibly stuck USB read channel. [2217]
[2025-04-17 10:57:48.143] WARNING[1973633]: chan_usbradio.c:2051
usbradio_read: USB read channel [2217] was not stuck.
[2025-04-17 10:57:51.214] ERROR[1973633]: chan_usbradio.c:1962
usbradio_read: Channel 2217: HID process has died or is not responding.
[2025-04-17 10:58:11.856] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File /home/kyle/pcr1000 does not exist in any format
[2025-04-17 10:58:11.856] WARNING[1973640][C-0000001f]: file.c:1322
ast_streamfile: Unable to open /home/kyle/pcr1000 (format (ulaw)):
Permission denied
[2025-04-17 10:58:11.856] WARNING[1973640][C-0000001f]: app_playback.c:520
playback_exec: Playback failed on IAX2/pi4-3040-4-2519 for
/home/kyle/pcr1000
[2025-04-17 10:58:11.904] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/1 does not exist in any format
[2025-04-17 10:58:12.818] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/6 does not exist in any format
[2025-04-17 10:58:13.701] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/2 does not exist in any format
[2025-04-17 10:58:14.452] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/letters/dot does not exist in any format
[2025-04-17 10:58:15.195] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/5 does not exist in any format
[2025-04-17 10:58:16.020] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/5 does not exist in any format
[2025-04-17 10:58:16.843] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/digits/0 does not exist in any format
[2025-04-17 10:58:17.721] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/letters/n does not exist in any format
[2025-04-17 10:58:18.342] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/letters/f does not exist in any format
[2025-04-17 10:58:18.928] WARNING[1973640][C-0000001f]: file.c:825
ast_openstream_full: File custom/letters/m does not exist in any format
[2025-04-17 10:58:19.539] WARNING[1973640][C-0000001f]: app_rpt.c:6366
rpt_exec: Cannot find specified system node 2216

-----------------------------------------------

3040-4*CLI> rpt restart
[2025-04-17 11:01:40.636] WARNING[1973696]: app_rpt.c:5789 rpt_master:
rpt_thread restarted on node 2216
[2025-04-17 11:01:40.637] WARNING[1973696]: app_rpt.c:5789 rpt_master:
rpt_thread restarted on node 2217
[2025-04-17 11:01:40.644] NOTICE[1973925]: chan_usbradio.c:2560
usbradio_setoption: Channel 2216: Set option TONE VERIFY, mode: OFF(0).
[2025-04-17 11:01:40.646] NOTICE[1973926]: chan_usbradio.c:2560
usbradio_setoption: Channel 2217: Set option TONE VERIFY, mode: OFF(0).
[2025-04-17 11:01:40.718] ERROR[1973928]: chan_usbradio.c:1597 setformat:
Channel 2217: Unable to open DSP device 0: Device or resource busy.
[2025-04-17 11:01:40.731] WARNING[1973926]: chan_usbradio.c:2040
usbradio_read: Possibly stuck USB read channel. [2217]
[2025-04-17 11:01:40.755] WARNING[1973926]: chan_usbradio.c:2051
usbradio_read: USB read channel [2217] was not stuck.
[2025-04-17 11:01:44.210] ERROR[1973926]: chan_usbradio.c:1962
usbradio_read: Channel 2217: HID process has died or is not responding.
[2025-04-17 11:01:44.224] ERROR[1973925]: chan_usbradio.c:1962
usbradio_read: Channel 2216: HID process has died or is not responding.
[2025-04-17 11:01:45.270] WARNING[1973932]: file.c:825 ast_openstream_full:
File custom/letters/a does not exist in any format
[2025-04-17 11:01:45.887] WARNING[1973932]: file.c:825 ast_openstream_full:
File custom/letters/p does not exist in any format
[2025-04-17 11:01:46.531] WARNING[1973932]: file.c:825 ast_openstream_full:
File custom/letters/d does not exist in any format
[2025-04-17 11:01:47.252] WARNING[1973932]: file.c:825 ast_openstream_full:
File custom/letters/i does not exist in any format
[2025-04-17 11:01:47.929] WARNING[1973932]: file.c:825 ast_openstream_full:
File custom/letters/s does not exist in any format

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>