AllStarLink/ASL3 #142: WARNING[1871435]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-174269273

kyle470, 2025-04-13:
**Describe the bug**
A clear and concise description of what the bug is.
Autopatch not working on latest update of ASL3

my rpt.conf function that I have used for years. Does not matter if I use quiet=1 or 0 etc:
69=autopatchup,context=icom-ctl,noct=1,farenddisconnect=1,dialtime=90000,quiet=1

[2025-04-13 17:03:21.775] WARNING[1871435]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-174269273
[2025-04-13 17:03:21.776] WARNING[1871436][C-00000017]: pbx.c:2936 pbx_extension_helper: No application 'Wait,1' for extension (icom-ctl, *11, 1)
[2025-04-13 17:03:21.786] WARNING[1871435]: app_rpt.c:1391 rpt_call:  has no PBX?
[2025-04-13 17:03:21.807] WARNING[1871435]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-174269273

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Software versions (listed in asl-menu, option 4)**
 - asl3: [e.g. 3.1.1-1]
 - asl3-asterisk: [e.g. 2:20.8.1+asl3-3.0.0-1]

**Have you run a software update and rebooted?**
 - YES or NO

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Platform

**Additional context**
Add any other context about the problem here.

mkmer, 2025-04-13:
The format of extensions.conf has changed with the Asterisk 20+ .  It looks like you have a malformed `Wait()` in your dialplan.

kyle470, 2025-04-14:
> exten => ${NODE},1,Ringing()

thanks, that was in a custom script, I've got the format fixed. I ignored the error as I am also getting errors such as these when enabling the Autopatch. The audio plays although I see these errors. I have also tried autopatch on another node without that custom script and do not get any errors besides "cannot play tone".

[2025-04-13 18:03:20.898] WARNING[1872815]: file.c:825 ast_openstream_full: File custom/letters/a does not exist in any format
[2025-04-13 18:03:21.515] WARNING[1872815]: file.c:825 ast_openstream_full: File custom/letters/p does not exist in any format
[2025-04-13 18:03:22.157] WARNING[1872815]: file.c:825 ast_openstream_full: File custom/letters/e does not exist in any format
[2025-04-13 18:03:22.822] WARNING[1872815]: file.c:825 ast_openstream_full: File custom/letters/n does not exist in any format
[2025-04-13 18:03:23.442] WARNING[1872815]: file.c:825 ast_openstream_full: File custom/letters/a does not exist in any format

mkmer, 2025-04-14:
https://github.com/AllStarLink/app_rpt/issues/587

kyle470, 2025-04-14:
I don't understand why this issue was closed,  the bug still exists, and I cannot use autopatch on multiple ASL3 boxes

Allan-N, 2025-04-14:
Also, have you looked at [ASL3 Manual : Autopatch](https://allstarlink.github.io/adv-topics/autopatch/?h=autopatch)

kyle470, 2025-04-14:
Hello. Thanks for pointing out the manual, I found a couple things that needed updating. I am now getting farther than before, although autopatch still does not work on the 3 Dell Wyse 3040 units I have tried thus far. The call now actually sets up and rings the SIP phone, however there is no audio either way, and the ASL3 box becomes unresponsive until I restart Asterisk.

Here is the output when attempting to make a call to a SIP phone on my LAN (across an IAX2 trunk). I have been able to call my PBX phones since I began using app_rpt in roughly 2005, so I know the PBX side of things is ok, pre-ASL3 anyway.

One thing I noticed right away is that the calls seem to be changing codecs, although I am only allowing ulaw on the trunk the call routes over. No idea where the Ooh or slin came from.

[2025-04-14 17:35:52.160] DEBUG[2209855][C-00000020]: chan_dahdi.c:9408 dahdi_indicate: Requested indication 3 on channel DAHDI/pseudo-457799289
[2025-04-14 17:35:52.161] DEBUG[2209855][C-00000020]: channel.c:4654 indicate_data_internal: Driver for channel 'DAHDI/pseudo-457799289' does not support indication 3, emulating it
[2025-04-14 17:35:52.161] DEBUG[2209855][C-00000020]: channel.c:5763 set_format: Channel DAHDI/pseudo-457799289 setting write format path: slin -> ulaw

[2025-04-14 17:35:57.297] DEBUG[2208946][C-00000020]: chan_iax2.c:10788 socket_process_helper: Ooh, voice format changed to 'ulaw'

**Running a debug on the PBX and only see a single call but ASL3 reports:**
[2025-04-14 17:35:51.241] DEBUG[2209855][C-00000020]: chan_iax2.c:2727 peercnt_add: ip callno count incremented to 2 for 192.168.2.17

[2025-04-14 17:35:51.242] DEBUG[2209855][C-00000020]: rtp_engine.c:2449 ast_rtp_instance_early_bridge_make_compatible: Can't find native functions for channel 'IAX2/pi4-3040-2-3915'
[2025-04-14 17:35:51.242] DEBUG[2209855][C-00000020]: channel.c:5763 set_format: Channel IAX2/pi4-3040-2-3915 setting read format path: ulaw -> ulaw

[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: stasis.c:620 stasis_topic_create_with_detail: Topic 'bridge:all/bridge:6de6fd75-976c-425b-b4d7-f35c63803ac8': 0x7f55e80113a0 created
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge_native_rtp.c:773 native_rtp_bridge_compatible: Bridge '6de6fd75-976c-425b-b4d7-f35c63803ac8' can not use native RTP bridge as two channels are required
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:531 find_best_technology: Bridge technology native_rtp is not compatible with properties of existing bridge.
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: dahdi/bridge_native_dahdi.c:705 native_bridge_compatible: Bridge 6de6fd75-976c-425b-b4d7-f35c63803ac8: Cannot use native DAHDI.  Must have two channels.
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:531 find_best_technology: Bridge technology native_dahdi is not compatible with properties of existing bridge.
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:521 find_best_technology: Bridge technology holding_bridge does not have any capabilities we want.
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:526 find_best_technology: Bridge technology softmix has less preference than simple_bridge (10 <= 50). Skipping.
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:546 find_best_technology: Chose bridge technology simple_bridge
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:792 bridge_base_init: Bridge 6de6fd75-976c-425b-b4d7-f35c63803ac8: calling simple_bridge technology constructor
[2025-04-14 17:35:57.552] DEBUG[2209855][C-00000020]: bridge.c:800 bridge_base_init: Bridge 6de6fd75-976c-425b-b4d7-f35c63803ac8: calling simple_bridge technology start

Entire CLI debug:
=========================================================================
Connected to Asterisk 22.2.0+asl3-3.3.0-6.deb12 currently running on 3040-2 (pid = 2208903)
[2025-04-14 17:35:40.816] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:41.452] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:42.090] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:42.461] DEBUG[2208957]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:42.728] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:43.364] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:44.002] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:44.638] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:44.700] DEBUG[2208957]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:45.275] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:35:46.230] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[... cut ...]
[2025-04-14 17:36:08.106] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:36:08.742] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:36:09.379] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
[2025-04-14 17:36:10.334] DEBUG[2208960]: chan_dahdi.c:9310 my_dahdi_write: Write returned -1 (Resource temporarily unavailable) on channel -2
3040-2*CLI>

mkmer, 2025-04-15:
I'm thinking something is wrong/old with your config files.  The "easiest way" to address this is to start with a clean set of configuration files (the ones included in the install), then set up using those files.  There are quite a few things that have changed since ASL2, and some are not obvious.

mkmer, 2025-04-15:
I'm curios if you are using pjsip or the old sip channel?  

I did get my autopatch to crash on a dialplan that was missing a `hangup()` step - I'm not sure if this is part of your problem, but worth looking at.

```
[voipms-outbound]
exten => _1NXXNXXXXXX,1,Dial(PJSIP/${EXTEN}@voipms)
same => n,Hangup()
```

kyle470, 2025-04-15:
I spun up a new build of ASL3 on my spare Pi4 last night and am pretty
close to being able to test autopatch, should be able to try tonight after
work

mkmer, 2025-04-16:
I have been able to crash the autopatch by having  a dialplan which ends quickly. It will work if the other side doesn't hang up to quickly.  I've worked up some fixes around addressing "fast" hangups (which include a bad dialplan, or no dialplan).
Having an autopatch context non existent int the dialplan will also crash... 

The crash should be address in the next release (Mayish), but in the mean time, it "should" work if the dialplan is working.
Of course your situation may be different from the crash I was able to generate, so I'll leave this open for now.

kyle470, 2025-04-16:
OK, I have got autopatch somewhat working on my new Pi4 test box but have noticed a new issue. VOX does not seem to work when connected to my simplex node. I have even tried voxalways=1 but when muting the PBX phone, PTT on my
 simplex node is still asserted.  This may be due to the fact I am using a hub node on this test box? I would think voxalways should still work even if the hub node is essentially full-duplex?

;;;;; Autopatch Commands ;;;;;
; Note, This may be a good place for other 2 digit frequently used commands
61 = autopatchup,noct = 1,context=autopatch,farenddisconnect = 1,dialtime = 20000,voxalways = 1,quiet=1  ; Autopatch up
0 = autopatchdn                                                 ; Autopatch down

[1999](node-main)
;;;;;;;;;;;;;;;;;;; Your node settings here ;;;;;;;;;;;;;;;;;;;
rxchannel = dahdi/pseudo            ; No radio (hub)
context = autopatch
callerid = "Allstar" <1999>  ; callerid for phone calls
duplex = 1
voxtimeout = 10000     ; VOX timeout time in ms
voxrecover = 2000      ; VOX recover time in ms
simplexpatchdelay = 25 ; Delay for transmit while in patch in 20ms increments
simplexphonedelay = 25 ; Delay for phone while in patch for 20ms increments

Also seeing the same codec oddities on this fresh install:
2025-04-16 11:39:15.269] DEBUG[29565][C-0000000e]: chan_iax2.c:10788 socket_process_helper: Ooh, voice format changed to 'ulaw'

Ooh is not even being loaded or running:
noload  = chan_ooh323.so                 ; Objective Systems H323 Channel

pi4test*CLI> module show like ooh
Module                         Description                              Use Count  Status      Support Level
0 modules loaded

---------------------------

Complete debug output for a working autopatch call with VOX not working:

pi4test*CLI> rpt fun 1999 *61502
[2025-04-16 11:39:10.163] DEBUG[29575]: app_rpt.c:1525 collect_function_digits: @@@@ action: autopatchup, param = noct = 1,context=autopatch,farenddisconnect = 1,dialtime = 20000,voxalways = 1,quiet=1
[2025-04-16 11:39:10.163] DEBUG[29575]: app_rpt.c:1531 collect_function_digits: @@@@ table index i = 1
[2025-04-16 11:39:10.163] DEBUG[29575]: app_rpt/rpt_functions.c:906 function_autopatchup: @@@@ Autopatch up
[2025-04-16 11:39:10.163] DEBUG[29600][C-0000000e]: chan_dahdi.c:14000 dahdi_request: Using channel -2
[2025-04-16 11:39:10.164] DEBUG[29600][C-0000000e]: stasis.c:620 stasis_topic_create_with_detail: Topic 'channel:1744821550.14': 0x7f90007b48 created
[2025-04-16 11:39:10.164] DEBUG[29600][C-0000000e]: channel.c:949 __ast_channel_alloc_ap: Channel 0x7f90001528 'DAHDI/pseudo-1324896632' allocated
[2025-04-16 11:39:10.164] DEBUG[29600]: app_rpt.c:1220 rpt_call: Requested channel DAHDI/pseudo-1324896632
[2025-04-16 11:39:10.164] DEBUG[29600]: chan_dahdi.c:9408 dahdi_indicate: Requested indication -1 on channel DAHDI/pseudo-1324896632
[2025-04-16 11:39:10.164] DEBUG[29600][C-0000000f]: chan_dahdi.c:14000 dahdi_request: Using channel -2
[2025-04-16 11:39:10.164] DEBUG[29600][C-0000000f]: stasis.c:620 stasis_topic_create_with_detail: Topic 'channel:1744821550.15': 0x7f900042f8 created
[2025-04-16 11:39:10.164] DEBUG[29600][C-0000000f]: channel.c:949 __ast_channel_alloc_ap: Channel 0x7f90009f48 'DAHDI/pseudo-1519270778' allocated
[2025-04-16 11:39:10.165] DEBUG[29600]: app_rpt.c:1237 rpt_call: Requested channel DAHDI/pseudo-1519270778
[2025-04-16 11:39:10.165] DEBUG[29600]: chan_dahdi.c:9408 dahdi_indicate: Requested indication -1 on channel DAHDI/pseudo-1519270778
[2025-04-16 11:39:10.542] WARNING[29600]: app_rpt/rpt_bridging.c:567 rpt_play_tone: Cannot start tone on DAHDI/pseudo-1519270778
[2025-04-16 11:39:10.543] DEBUG[29601][C-0000000e]: pbx.c:2947 pbx_extension_helper: Launching 'Dial'
    -- Executing [502@autopatch:1] Dial("DAHDI/pseudo-1324896632", "IAX2/pi4-pitest/502") in new stack
[2025-04-16 11:39:10.543] DEBUG[29601][C-0000000e]: app_dial.c:2366 dial_exec_full:  DAHDI/pseudo-1324896632: Data: IAX2/pi4-pitest/502
[2025-04-16 11:39:10.543] DEBUG[29601][C-0000000e]: chan_iax2.c:2727 peercnt_add: ip callno count incremented to 3 for 192.168.2.17
[2025-04-16 11:39:10.543] DEBUG[29601][C-0000000e]: stasis.c:620 stasis_topic_create_with_detail: Topic 'channel:1744821550.16': 0x7f980042f8 created
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: channel.c:949 __ast_channel_alloc_ap: Channel 0x7f980070f8 'IAX2/pi4-pitest-5002' allocated
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: rtp_engine.c:2449 ast_rtp_instance_early_bridge_make_compatible: Can't find native functions for channel 'IAX2/pi4-pitest-5002'
    -- Called IAX2/pi4-pitest/502
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: channel.c:5763 set_format: Channel IAX2/pi4-pitest-5002 setting read format path: ulaw -> ulaw
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: channel.c:5763 set_format: Channel DAHDI/pseudo-1324896632 setting write format path: ulaw -> ulaw
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: channel.c:5763 set_format: Channel DAHDI/pseudo-1324896632 setting read format path: ulaw -> ulaw
[2025-04-16 11:39:10.544] DEBUG[29601][C-0000000e]: channel.c:5763 set_format: Channel IAX2/pi4-pitest-5002 setting write format path: ulaw -> ulaw
[2025-04-16 11:39:10.544] DEBUG[29545]: stasis.c:620 stasis_topic_create_with_detail: Topic 'devicestate:all/IAX2/pi4-pitest': 0x7f700016f8 created
    -- Call accepted by 192.168.2.17:4569 (format ulaw)
    -- Format for call is (ulaw)
[... cut ...]
[2025-04-16 11:40:04.649] DEBUG[29564]: chan_iax2.c:2727 peercnt_add: ip callno count incremented to 3 for 192.168.2.17
[2025-04-16 11:40:04.650] DEBUG[29570]: chan_iax2.c:3127 sched_delay_remove: schedule decrement of callno used for 192.168.2.17 in 60 seconds
[2025-04-16 11:40:04.650] DEBUG[29570]: chan_iax2.c:11355 socket_process_helper: Peer pi4-pitest: got pong, lastms 1, historicms 1, maxms 2000
pi4test*CLI>
[2025-04-16 11:40:30.576] DEBUG[29595]: manager.c:7040 process_message: Running action 'Command'
pi4test*CLI>
Disconnected from Asterisk server

ve7fet, 2025-07-13:
It also appears there is no default `/etc/dahdi/system.conf` installed with ASL3, which would load and set the default country to `us` (but `tonezone` in `/etc/asterisk/dahdi.conf` still likely needs to be set too).

https://github.com/asterisk/dahdi-tools/blob/master/system.conf.sample

kyle470, 2025-07-13:
Hello. Guessing you're referring to /etc/asterisk/chan_dahdi.conf. I don't have
etc/asterisk/dadhi.conf

I tried setting the tonezone to 0 and setting quiet=0 for the autopatch. No luck. 

I do have my autopatch working now by using voxalways=1,quiet=1

73

Kyle
K0KN

ve7fet, 2025-07-13:
> Hello. Guessing you're referring to /etc/asterisk/chan_dahdi.conf. I don't have etc/asterisk/dadhi.conf
> 
> I tried setting the tonezone to 0 and setting quiet=0 for the autopatch. No luck.
> 
> I do have my autopatch working now by using voxalways=1,quiet=1
> 
> 73
> 
> Kyle K0KN

Ahh yes indeed, meant `/etc/asterisk/chan_dadhi.conf`, updated my post.

I expect you probably don't have an `/etc/dadhi/system.conf` file. You could try creating one with the contents:

```
loadzone = us
defaultzone=us
```

And restart Asterisk (of course). Those seem to be the only two lines actually enabled from the sample file at https://github.com/asterisk/dahdi-tools/blob/master/system.conf.sample and they are probably pertinent to this issue?

Lee

ve7fet, 2025-07-14:
Well boo, and here I thought I might have stumbled onto the solution. ;) I guess we'll have to wait for the smart people in the room to have a further look.

Lee

mkmer, 2025-07-18:
`tonezone` is set via code here:
https://github.com/AllStarLink/app_rpt/blob/2c721d13d13fcc39b10123eac9427d72d5957564/apps/app_rpt.c#L1299
based on the config files (if we have one):
https://github.com/AllStarLink/app_rpt/blob/2c721d13d13fcc39b10123eac9427d72d5957564/apps/app_rpt/rpt_config.c#L868

Of course it's not in the stock rpt.conf file either :(

Interestingly, I get the call terminated message regardless of `quiet = 1` or `0`.
Still looking at "why" we can't play on the `genchannel`  (It appears this is the "spot" where things go wrong with quiet=0)
https://github.com/AllStarLink/app_rpt/blob/2c721d13d13fcc39b10123eac9427d72d5957564/apps/app_rpt.c#L1306-L1310 

If you make a call all in one (aka *611234567890) it "should" skip the dialtone generation.  If this "works" , maybe we get another error that will help point me in the right direction as to what is broken here.