AllStarLink/ASL3 #156: extensions.conf: NODE Variable Breaks Contexts With More Than One Node on Server

ve7fet, 2025-08-20:
**Describe the bug**
When a node is created, `node-setup` sets the `[global]` variable `NODE=<nodenumber>` in `extensions.conf`.

This is all fine, when there is a single node on the server. However, when you add another node to the sever, that breaks contexts in `extensions.conf`.

Case in point, `[iaxrpt]`. If you have two nodes running on the box, you can only connect with IAXRpt (or any IAX client) to the first node that was setup (`NODE=<nodenumber>`). Attempting to connect to any other node on the server **fails silently** (no messages on the CLI).

This is because the stock context is:

```
[iaxrpt]
; Entered from iaxrpt in iax.conf
; Info: The X option passed to the Rpt application
; disables the normal security checks.
; Because incoming connections are validated in iax.conf,
; and we don't know where the user will be coming from in advance,
; the X option is required.
exten => ${NODE},1,rpt(${EXTEN},X)       ; NODE is the Name field in iaxrpt
```

So, when user `iaxrpt` connects, only the `NODE` global is passed to this context. If you are trying to connect to another node number running on the server, there is no matching extension for that in this context, so it just fails silently.

We need a better way to handle multiple nodes.

A workaround for this is to add another line to that context with the other node number(s):

```
[iaxrpt]
; Entered from iaxrpt in iax.conf
; Info: The X option passed to the Rpt application
; disables the normal security checks.
; Because incoming connections are validated in iax.conf,
; and we don't know where the user will be coming from in advance,
; the X option is required.
exten => ${NODE},1,rpt(${EXTEN},X)       ; NODE is the Name field in iaxrpt
exten => <secondnodenumber>,1,rpt(${EXTEN},X)       ; replace <secondnodenumber> with the actual additional node number
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add a second node to the server
2. Configure an external application/client to connect to the second node number
3. Fail silently (unable to connect, no messages in CLI)

**Expected behavior**
Be able to connect to any provisioned node number on the server with an external client.

**Software versions (listed in asl-menu, option 4)**
```
********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.12.34+rpt-rpi-v8

Asterisk      : 22.4.1+asl3-3.5.5-1.deb12
ASL [app_rpt] : 3.5.5

Installed ASL packages :

  Package                         Version
  ==============================  ==============================
  allmon3                         1.6.0-3.deb12
  asl3                            3.8-1.deb
  asl3-asterisk                   2:22.4.1+asl3-3.5.5-1.deb12
  asl3-asterisk-config            2:22.4.1+asl3-3.5.5-1.deb12
  asl3-asterisk-modules           2:22.4.1+asl3-3.5.5-1.deb12
  asl3-menu                       1.14-1.deb12
  asl3-pi-appliance               1.10.0-1.deb12
  asl3-update-nodelist            1.5.1-1.deb12
  cockpit                         287.1-0+deb12u3
  cockpit-bridge                  287.1-0+deb12u3
  cockpit-networkmanager          287.1-0+deb12u3
  cockpit-packagekit              287.1-0+deb12u3
  cockpit-pcp                     287.1-0+deb12u3
  cockpit-sosreport               287.1-0+deb12u3
  cockpit-storaged                287.1-0+deb12u3
  cockpit-system                  287.1-0+deb12u3
  cockpit-wifimanager             1.1.1-1.deb12
  cockpit-ws                      287.1-0+deb12u3
  dahdi                           1:3.1.0-2
  dahdi-dkms                      1:3.4.0-7.asl
  dahdi-linux                     1:3.4.0-7.asl
```

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
  - :Platform Pi 3

InterLinked1, 2025-08-20:
> [@InterLinked1](https://github.com/InterLinked1) Any thoughts on how to changes to `extensions.conf` to support multiple nodes?
> 
> * Can we avoid having one line per node in the file?

Most likely.

> * Is there be a way to avoid having to add/remove lines as nodes are added/removed?

One option could be to use a pattern, e.g.

`exten => _X!,` to match all nodes by default, but then you could add specific ones to override, e.g. `exten => 2001,`,

I think a good idea might be to have a function in app_rpt that returns if a node exists, e.g. `exten => _X!,1,NoOp(${RPT_NODE(${EXTEN},exists)})`

Then you could have a pattern to match any extension, and have an ExecIf that bails out if the node doesn't exist and calls `Rpt` otherwise, or whatever you want to do. Would that accomplish the goal?

> * Is there be a way to have `app_rpt` set a [global] variable with a list of nodes that the dialplan could use?

There should be no need for this, and this should not be done. I don't see the need for global variables in the above example anyways.

nostar, 2025-11-21:
It is the case on current ASL3.  I have 2 ASL3 servers.  One is hosting a single node, and allow/deny lists work as expected.  The other ASL3 server is hosting 2 nodes, and when node B has a deny list, then node A ignores it's allow list.  For example, I add an allow list for node A with one callsign or node ID.  If a deny list exists for node B, then the allow list for node A is ignored and anyone can connect to node A even though it is supposed to be restricted to one callsign or node ID.

nostar, 2025-11-22:
I'm pretty sure I didn't make any manual changes, but here's the file anyway
```
asl@node501680:~ $ cat /etc/asterisk/extensions.conf
[general]

static = yes       ; These two lines prevent the command-line interface
writeprotect = yes ; from overwriting the config file. Leave them here.

[globals]
HOMENPA = 999 ; change this to your Area Code
NODE = 501680   ; change this to your node number

[default]
exten => i,1,Hangup

[radio-secure]
;exten => ${NODE},1,rpt(${EXTEN})
exten => _XXXX!,1,NoOp(Connect from node: ${CALLERID(num)})
	same => n,NoOp(Connect to: ${EXTEN})
	;same => n,NoOp(The IAXPEER is ${IAXPEER(CURRENTCHANNEL)})
	same => n,NoOp(The Channel IP is ${CHANNEL(peerip)})

	;If channel IP eq localhost go to connect
	same => n,GotoIf($["${CHANNEL(peerip)}" = "127.0.0.1"]?connect)

	;If allowlist/extension dosen't exist check the denylist
	same => n,GotoIf($[${DB_KEYCOUNT(allowlist/${EXTEN})} = 0]?denylist) ;goto denylist
	;If allowlist/extension/callerid exists go to connect
	same => n,GotoIf(${DB_EXISTS(allowlist/${EXTEN}/${CALLERID(num)})}?connect) ;goto connect
	same => n,NoOp(${EXTEN} not in allowlist, Hangup)
	same => n,Hangup

	;If denylist/extension/callerID doesn't exist goto connect
	same => n(denylist),GotoIf(${DB_EXISTS(denylist/${EXTEN}/${CALLERID(num)})}?:connect) ;don't goto connect
	same => n,NoOp(${EXTEN} is in denylist, Hangup)
	same => n,Hangup

	same => n(connect),rpt(${EXTEN})
	same => n,Hangup
		
[iaxrpt]
; Entered from iaxrpt in iax.conf
; Info: The X option passed to the Rpt application
; disables the normal security checks.
; Because incoming connections are validated in iax.conf,
; and we don't know where the user will be coming from in advance,
; the X option is required.
exten => ${NODE},1,rpt(${EXTEN}|X)       ; NODE is the Name field in iaxrpt

[iax-client]                            ; for IAX VoIP clients.
exten => ${NODE},1,Ringing()
	same => n,Wait(1)
	same => n,Answer()
	same => n,Set(CALLSIGN=${CALLERID(name)})
	same => n,NoOp(Caller ID name is ${CALLSIGN})
	same => n,NoOp(Caller ID number is ${CALLERID(number)})
	same => n,GotoIf(${ISNULL(${CALLSIGN})}?hangit)
	same => n,Playback(rpt/connected-to&rpt/node)
	same => n,SayDigits(${NODE})
	same => n,rpt(${NODE}|P|${CALLSIGN}-P)
	same => n(hangit),NoOp(No Caller ID Name)
	same => n,Playback(connection-failed)
	same => n,Wait(1)
	same => n,Hangup

; Comment-out the following clause if you want Allstar Autopatch service
[pstn-out]
exten => _NXXNXXXXXX,1,playback(ss-noservice)
	same => n,Congestion

; Un-comment out the following clause if you want Allstar Autopatch service
;[pstn-out]
;exten => _NXXNXXXXXX,1,Dial(IAX2/allstar-autopatch/\${EXTEN})
; same => n,Busy

[invalidnum]
exten => s,1,Wait(3)
	same => n,Playback(ss-noservice)
	same => n,Wait(1)
	same => n,Hangup

[radio]
exten => _X11,1,Goto(check_route,${EXTEN},1);
exten => _NXXXXXX,1,Goto(check_route,1${HOMENPA}${EXTEN},1)
exten => _1XXXXXXXXXX,1,Goto(check_route,${EXTEN},1)
exten => _07XX,1,Goto(parkedcalls,${EXTEN:1},1)
exten => 00,1,Goto(my-ip,s,1)

[check_route]
exten => _X.,1,Noop(${EXTEN})
; no 800
exten => _1800NXXXXXX,2,Goto(invalidnum,s,1)
exten => _1888NXXXXXX,2,Goto(invalidnum,s,1)
exten => _1877NXXXXXX,2,Goto(invalidnum,s,1)
exten => _1866NXXXXXX,2,Goto(invalidnum,s,1)
exten => _1855NXXXXXX,2,Goto(invalidnum,s,1)
; no X00 NPA
exten => _1X00XXXXXXX,2,Goto(invalidnum,s,1)
; no X11 NPA
exten => _1X11XXXXXXX,2,Goto(invalidnum,s,1)
; no X11
exten => _X11,2,Goto(invalidnum,s,1)
; no 555 Prefix in any NPA
exten => _1NXX555XXXX,2,Goto(invalidnum,s,1)
; no 976 Prefix in any NPA
exten => _1NXX976XXXX,2,Goto(invalidnum,s,1)
; no NPA=809
exten => _1809XXXXXXX,2,Goto(invalidnum,s,1)
; no NPA=900
exten => _1900XXXXXXX,2,Goto(invalidnum,s,1)

; okay, route it
exten => _1NXXXXXXXXX,2,Goto(pstn-out,${EXTEN:1},1)
exten => _X.,2,Goto(invalidnum,s,1)

[my-ip]
exten => s,1,Wait(1)
	same => n,SayAlpha(${CURL(http://myip.vg)})
	same => n,Hangup

[allstar-sys]
exten => _1.,1,Rpt(${EXTEN:1}|Rrpt/node:NODE:rpt/in-call:digits/0:PARKED|120)
exten => _1.,n,Hangup

exten => _2.,1,Ringing
exten => _2.,n,Wait(3)
exten => _2.,n,Answer
exten => _2.,n,Playback(rpt/node)
exten => _2.,n,Saydigits(${EXTEN:1})
exten => _2.,n,Rpt(${EXTEN:1}|P|${CALLERID(name)}-P)
exten => _2.,n,Hangup

exten => _3.,1,Ringing
exten => _3.,n,Wait(3)
exten => _3.,n,Answer
exten => _3.,n,Playback(rpt/node)
exten => _3.,n,Saydigits(${EXTEN:1})
exten => _3.,n,Rpt(${EXTEN:1}|Pv|${CALLERID(name)}-P)
exten => _3.,n,Hangup

exten => _4.,1,Ringing
exten => _4.,n,Wait(3)
exten => _4.,n,Answer
exten => _4.,n,Playback(rpt/node)
exten => _4.,n,Saydigits(${EXTEN:1})
exten => _4.,n,Rpt(${EXTEN:1}|D|${CALLERID(name)}-P)
exten => _4.,n,Hangup

exten => _5.,1,Ringing
exten => _5.,n,Wait(3)
exten => _5.,n,Answer
exten => _5.,n,Playback(rpt/node)
exten => _5.,n,Saydigits(${EXTEN:1})
exten => _5.,n,Rpt(${EXTEN:1}|Dv|${CALLERID(name)}-P)
exten => _5.,n,Hangup

[allstar-public]

exten => s,1,Ringing
	same => n,Set(RESP=${CURL(https://register.allstarlink.org/cgi-bin/authwebphone.pl?${CALLERID(name)})})
	same => n,Set(NODENUM=${CALLERID(number)})
	same => n,GotoIf($["${RESP:0:1}" = "?"]?hangit)
	same => n,GotoIf($["${RESP:0:1}" = ""]?hangit)
	same => n,GotoIf($["${RESP:0:5}" != "OHYES"]?hangit)
	same => n,Set(CALLSIGN=${RESP:5})
	;exten => s,n,GotoIf($[${DB_EXISTS(whitelist/${CALLSIGN})}]?:hangit)  ; Not in whitelist
	same => n,Wait(3)
	same => n,Playback(rpt/node,noanswer)
	same => n,Saydigits(${NODENUM})
	same => n,Set(CALLERID(name)=${CALLSIGN})
	same => n,Set(CALLERID(num)=0)
	same => n,Rpt(${NODENUM}|X)
	same => n,Hangup
	same => n(hangit),Answer
	same => n,Wait(1)
	same => n,Hangup

#tryinclude custom/extensions.conf
```

nostar, 2025-11-22:
It seems to be more broken than originally described.  Callsigns dont work in the allow list at all, for either node.  I put one callsign in the allow list for one or both nodes, but any callsign connects fine.  All of this works correctly on the server with a single node, though.

Allan-N, 2025-11-22:
>Callsigns dont work in the allow list at all, for either node

That's because you don't have the latest version of the `extensions.conf` file where we added WT callsign support.  The current version of the file is [GitHub: app_rpt/configs/rpt/extensions.conf](https://github.com/AllStarLink/app_rpt/blob/master/configs/rpt/extensions.conf)

Allan-N, 2025-11-24:
> I think this file fixed the issue.

Great.  If so, please "Close" this issue.

Also, if you needed to make any changes to your `extensions.conf` file (besides updating the NODE= line) then please drop us a comment here.